RobotWebTools/rclnodejs #34: Figure the potential issue behind an endurance test

kenny-y, 2017-07-04:
Node.js client publisher: still working after 16,058,500 (`rcl_init` and `rcl_shutdown` between every 2 messages)
Node.js client subscription: stopped working after 1,629,680 messages (got stuck and can't take any new messages)
C++ client subscription (`examples_rclcpp_minimal_subscriber_lambda`): stopped working after 6,185,940 messages (got stuck and can't take any new messages)

Note: a new instance of subscription of C++ or Node.js can take new messages from the Node.js publisher.

kenny-y, 2017-07-04:
CC @minggangw

minggangw, 2017-09-06:
It sees that with the latest master branch of [ROS2.0](https://github.com/ros2/ros2/blob/master/ros2.repos), the stability is better now and more robust, so close this.

closed 2017-09-06 by minggangw