RobotWebTools/rclnodejs #39: Only 1 test passed

qiuzhong, 2017-07-19:
## Issue description
I tried to clone the repository and run `npm install` and then `npm test` but got only 1 test passed.
Notice: ros2-related native libraries already have been installed.

## Steps to reproduced
1. Install ros2 by following [this guide](https://github.com/ros2/ros2/wiki/Linux-Development-Setup)
2. `git clone https://github.com/RobotWebTools/rclnodejs.git && cd rclnodejs && npm install`
3. `npm install -g mocha && npm test`

The I got this:
```

> rclnodejs@0.0.1 test /home/qiuzhong/Workspace/Work/ros2/robotwebtools/rclnodejs
> mocha

  rclnodejs node test suite
    ✓ Try creating a node

  rclnodejs publisher test suite
    1) Try creating a publisher
    2) Try publish a message

  ROSIDL Node.js message generator test suite
    3) Try require all message classes
    4) Try use stdmsgs/msg/String.msg, using getMessageClass() function overrides
    5) Testing message with all-primitive members - ColorRGBA
    6) Testing copy-constructor - Duration
    7) Testing assignment of an all-primitive message - Time
    8) Testing a compound message - Pose
    9) Testing constants - GoalStatus
    10) Testing array - Int32

  1 passing (963ms)
  10 failing

  1) rclnodejs publisher test suite Try creating a publisher:
     TypeError: Cannot read property 'getMessageType' of undefined
      at Context.<anonymous> (test/test-publisher.js:33:32)

  2) rclnodejs publisher test suite Try publish a message:
     TypeError: Cannot read property 'getMessageType' of undefined
      at Context.<anonymous> (test/test-publisher.js:40:32)

  3) ROSIDL Node.js message generator test suite Try require all message classes:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at msgTypeList.forEach (test/test-rosidl-message-generator.js:37:35)
      at Array.forEach (native)
      at Context.<anonymous> (test/test-rosidl-message-generator.js:36:17)

  4) ROSIDL Node.js message generator test suite Try use stdmsgs/msg/String.msg, using getMessageClass() function overrides:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at Context.<anonymous> (test/test-rosidl-message-generator.js:42:34)

  5) ROSIDL Node.js message generator test suite Testing message with all-primitive members - ColorRGBA:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at Context.<anonymous> (test/test-rosidl-message-generator.js:62:33)

  6) ROSIDL Node.js message generator test suite Testing copy-constructor - Duration:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at Context.<anonymous> (test/test-rosidl-message-generator.js:76:33)

  7) ROSIDL Node.js message generator test suite Testing assignment of an all-primitive message - Time:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at Context.<anonymous> (test/test-rosidl-message-generator.js:97:33)

  8) ROSIDL Node.js message generator test suite Testing a compound message - Pose:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at Context.<anonymous> (test/test-rosidl-message-generator.js:119:33)

  9) ROSIDL Node.js message generator test suite Testing constants - GoalStatus:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at Context.<anonymous> (test/test-rosidl-message-generator.js:182:33)

  10) ROSIDL Node.js message generator test suite Testing array - Int32:
     TypeError: Cannot read property 'getMessageClass' of undefined
      at Context.<anonymous> (test/test-rosidl-message-generator.js:220:33)

npm ERR! Test failed.  See above for more details.

```

kenny-y, 2017-07-19:
Thanks for finding the issue. Please be noted that this repo is still in development and not ready for external use so far... we'll publish it to npmjs.com when it's ready.

Meanwhile, we'll look into it... (I guess we've been in feature branch for a while and missed the status change of master)

Honry, 2017-07-19:
@kenny-y, @qiuzhong comes from Web QA team responsible for test development for node.js RCL. This issue is found during ramping up, we are aware that this repo is still in development but thinking it's good to pose issue as early as possible. :)

kenny-y, 2017-07-19:
Okay, copy that... Thanks for the info.
I thought there was external users who wanted to use this repo in early stage...
Smile.jpg...

PR #44 (2017-08-07): [UT]Fix the failure due to the changing of JavaScript interfaces.
Fix #39
commit 5ebdde12 (2017-08-07): [UT]Fix the failure due to the changing of JavaScript interfaces

Fix #39
closed 2017-08-07 by minggangw