RobotWebTools/rclnodejs #55: Multiple nodes in one process

minggangw, 2017-08-07:


minggangw, 2017-10-12:
@qiuzhong would you please try to create two ros nodes in one programme to see if it works?

qiuzhong, 2017-10-13:
@minggangw , do you mean I create two ROS nodes (like a publisher and a subscription) in a JavaScript file and see if it can communicate well?

minggangw, 2017-10-13:
Yes, two nodes in a single js file, although I haven't seen that kind of usage before, please just try and see the result, thanks!

qiuzhong, 2017-10-13:
Just try it with the code:
```
const rclnodejs = require('./rclnodejs/index.js');

rclnodejs.init().then(() => {
    var nodePublisher = rclnodejs.createNode('publisher', '/my_ns');
    var nodeSubscription = rclnodejs.createNode('subscription', '/my_ns');

    const RclString = rclnodejs.require('std_msgs').msg.String;
    var msg = new RclString();
    msg.data = 'hello world';

    var publisher = nodePublisher.createPublisher(RclString, 'chatter');
    setInterval(() => {
        publisher.publish(msg);
    }, 1000);
    var subscription = nodeSubscription.createSubscription(RclString, 'chatter', (msg) => {
        console.log(`received: ${msg.data}`);
    });
    rclnodejs.spin(nodePublisher);
    rclnodejs.spin(nodeSubscription);
}).catch((err) => {
    console.log(err);
});
```

And it turns out that the code works quite well!

minggangw, 2017-10-13:
Great, thanks a lot, so let's close it.

closed 2017-10-13 by minggangw

mis-eu, 2020-12-11:
> Just tried it with the code:
> 
> ```
> const rclnodejs = require('./rclnodejs/index.js');
> 
> rclnodejs.init().then(() => {
>     var nodePublisher = rclnodejs.createNode('publisher', '/my_ns');
>     var nodeSubscription = rclnodejs.createNode('subscription', '/my_ns');
> 
>     const RclString = rclnodejs.require('std_msgs').msg.String;
>     var msg = new RclString();
>     msg.data = 'hello world';
> 
>     var publisher = nodePublisher.createPublisher(RclString, 'chatter');
>     setInterval(() => {
>         publisher.publish(msg);
>     }, 1000);
>     var subscription = nodeSubscription.createSubscription(RclString, 'chatter', (msg) => {
>         console.log(`received: ${msg.data}`);
>     });
>     rclnodejs.spin(nodePublisher);
>     rclnodejs.spin(nodeSubscription);
> }).catch((err) => {
>     console.log(err);
> });
> ```
> 
> And it turns out that the code works quite well!

Can the above be adapted for PoseStamped as an msg.Array and  RclArray ?

```
const rclnodejs = require('./node_modules/rclnodejs/index.js');

rclnodejs.init().then(() => {
    var nodePublisher = rclnodejs.createNode('publisher', '/goal');
    var nodeSubscription = rclnodejs.createNode('subscription', '/goal');

    const RclArray = rclnodejs.require('geometry_msgs').msg.Array;
    var msg = new RclArray();
    msg.data = {
        header: {
            stamp: {
              secs: 0,
              nsecs: 100
            },
            frame_id: "world"
          },
          pose: {
            position: {
              x: 0.0,
              y: 0.0,
              z: 0.0
            },
            orientation: {
              x: 0.0,
              y: 0.0,
              z: 0.0,
              w: 1.0
            }
        }
    };

    var publisher = nodePublisher.createPublisher(RclArray, 'PoseStamped');
    setInterval(() => {
        publisher.publish(msg);
    }, 1000);
    var subscription = nodeSubscription.createSubscription(RclArray, 'PoseStamped', (msg) => {
        console.log(`received: ${msg.data}`);
    });
    rclnodejs.spin(nodePublisher);
    rclnodejs.spin(nodeSubscription);
}).catch((err) => {
    console.log(err);
});
```

ros2jsguy, 2020-12-11:
@mis-eu 
I don't think your example is valid due to your referencing a non-existent message type, i.e., geometry_msgs/msg/Array is not defined. See https://github.com/ros2/common_interfaces/tree/master/geometry_msgs/msg

If you want to send a message containing an array of PoseStampted you will need to define a custom message. Here's an example from the nav_msgs package: https://github.com/ros2/common_interfaces/blob/master/nav_msgs/msg/Path.msg. Here's a simple tutorial on defining custom messages: https://index.ros.org/doc/ros2/Tutorials/Custom-ROS2-Interfaces/

Hope this helps.