RobotWebTools/rclnodejs #58: Namespce of node/topic

minggangw, 2017-08-07:


minggangw, 2017-10-12:
@qiuzhong would you help to verify this on `rclnodejs`, you can reference https://github.com/ros2/rclpy/blob/22bc843c40ddb2f0ac79be69693484b18ca2a9b7/rclpy/test/test_node.py#L29 to see the usage, the interface of `rclnodejs` is similar with the one of `rclpy`, thanks!

qiuzhong, 2017-10-13:
@minggangw , sorry for the reply you late. 

Actually, the `rclnodejs` has already supported the namespace of node/topic. There are some cases in `test-node.js` cover this. However, some APIs like
* `node.get_namespace()`
* `node.get_name()`
in `rclpy` are not available in `rclnodejs`. I believe you'll implement them in M4.

minggangw, 2017-10-13:
I see, the `node.get_namespace()` and `node.get_name()` are recorded in #63, so I think we can close this one now.

qiuzhong, 2017-10-13:
Agreed. Please close it.

minggangw, 2017-10-13:
Thanks, and closing...

closed 2017-10-13 by minggangw