RobotWebTools/rclnodejs #63: Some methods not implemented

qiuzhong, 2017-08-14:
In contrast to [rclpy](https://github.com/ros2/rclpy), some methods are not implemented. So I failed to translate some rcpy test cases that call these methods to rclnodejs. Here is the list of these methods:

| Methods | Used in rclpy tests |
| --- | ---|
| rcl.node getter/setter | [TestNode.test_accessors](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_node.py#L39) |
| rcl.node.handler getter/setter | [TestNode.test_accessors](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_node.py#L39) |
| node.get_namespace() | [TestCreateNode.test_create_node_with_empty_namespace](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_create_node.py#L52) |
| node.get_name() | [TestNode.test_accessors](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_node.py#L40) |
| rcl.spin_once() | [test_timer_zero_callbacks10hertz](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_timer.py#L42) |
| rcl.validate_full_topic_name.validate_full_topic_name() | [TestValidateFullTopicName.test_validate_full_topic_name](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_full_topic_name.py#L32) |
| rcl.topic_or_service_is_hidden.topic_or_service_is_hidden() | [TestTopicOrServiceIsHidden.test_topic_or_service_is_hidden](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_topic_or_service_is_hidden.py#L44) |
| rcl.validate_node_name.validate_node_name() | [TestValidateNodeName.test_validate_node_name](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_node_name.py#L29) |
| rcl.test_validate_topic_name.test_validate_topic_name() | [TestValidateTopicName.test_validate_topic_name](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_topic_name.py#L32) |
| rcl.validate_namespace.validate_namespace() | [TestValidateNamespace.test_validate_namespace](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_namespace.py#L30) |

minggangw, 2017-08-22:
| Methods | Used in rclpy tests | Used in rclnodejs
| --- | ---| --- |
| rcl.node getter/setter | [TestNode.test_accessors](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_node.py#L39) | the handle is transparent to developers, and won't change once being assigned, only the getter method will offered. We can do some white-box testing. |
| rcl.node.handler getter/setter | [TestNode.test_accessors](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_node.py#L39) | same with above |
| node.get_namespace() | [TestCreateNode.test_create_node_with_empty_namespace](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_create_node.py#L52) | will add |
| node.get_name() | [TestNode.test_accessors](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_node.py#L40) | will add |
| rcl.spin_once() | [test_timer_zero_callbacks10hertz](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_timer.py#L42) | As we can't control the eventloop in JavaScript, we will drop this |
| rcl.validate_full_topic_name.validate_full_topic_name() | [TestValidateFullTopicName.test_validate_full_topic_name](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_full_topic_name.py#L32) |  will add |
| rcl.topic_or_service_is_hidden.topic_or_service_is_hidden() | [TestTopicOrServiceIsHidden.test_topic_or_service_is_hidden](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_topic_or_service_is_hidden.py#L44) |  will add |
| rcl.validate_node_name.validate_node_name() | [TestValidateNodeName.test_validate_node_name](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_node_name.py#L29) | will add |
| rcl.test_validate_topic_name.test_validate_topic_name() | [TestValidateTopicName.test_validate_topic_name](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_topic_name.py#L32) | will add |
| rcl.validate_namespace.validate_namespace() | [TestValidateNamespace.test_validate_namespace](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_validate_namespace.py#L30) | will add |

minggangw, 2017-10-16:
Start to work on this issue.

qiuzhong, 2017-10-16:
Great!

closed 2017-10-20 by minggangw

minggangw, 2017-10-20:
After 4be0126d has been merged, interfaces, excluded `rcl.node getter/setter`, `rcl.node.handler getter/setter` and`rcl.spin_once()`, should be implemented, please @qiuzhong verify it, thanks!

qiuzhong, 2017-10-20:
@minggangw , I'm writing case to cover them. If I encounter any problems, I will file them.

qiuzhong, 2017-10-20:
Verified. The missing rclnodejs APIs in the list above are complete and work well now! Test cases are in PR #150

minggangw, 2017-10-20:
Thanks a lot, I will finish the rest.

PR #147 (2017-10-20): [validator]Implement the validator
This patch implemented some functions of validator, new interfaces
include:
    validator.validateFullTopicName
    validator.validateNodeName
    validator.validateTopicName
    validator.validateNamespace

Other new interfaces include:
    node.name
    node.namespace
    rclnodejs.isTopicOrServiceHidden

There is an example to show the usage of validator, and the
corresponding document is updated.

Fix #63
commit 4be0126d (2017-10-20): [validator]Implement the validator

This patch implemented some functions of validator, new interfaces
include:
    validator.validateFullTopicName
    validator.validateNodeName
    validator.validateTopicName
    validator.validateNamespace

Other new interfaces include:
    node.name
    node.namespace
    rclnodejs.isTopicOrServiceHidden

There is an example to show the usage of validator, and the
corresponding document is updated.

Fix #63