RobotWebTools/rclnodejs #66: Double init() invoking raises just a warning

qiuzhong, 2017-08-15:
For `rclnodejs.init()` method, double invoking will raise a warning:
```
init() called
Warning - rclnodejs is already initialized. These arguments will be ignored: []
init() called again
```

My test code (put it in example directory):
```
const rclnodejs = require('../index.js');

rclnodejs.init().then(function() {
    console.log('init() called');
}).then(function() {    
    rclnodejs.init();
    console.log('init() called again');
}).catch(function(err) {
    console.log(err);
});

```

While for `rclnodejs.shutdown()`, double invoking will simply raise an Error.

minggangw, 2017-08-15:
We'd better uniform these two behaviours (`init` and `shutdown`) to make it consistent, thanks!

closed 2017-08-17 by minggangw

qiuzhong, 2017-08-21:
Verified. The `init` and `shutdown` methods now behave consistently now.

PR #69 (2017-08-17): Throw an error when double init/shutdown
Currently, the rclnodejs doesn't reject the promise when it's initialized
two times, and does throw an error when it was shut twice.

To make these two behaviors consistently, this patch implements a strategy,
which will throw an error when init/shutdown are called continuously.

Fix #66
commit 12b235c7 (2017-08-17): Throw an error when double init/shutdown

Currently, the rclnodejs doesn't reject the promise when it's initialized
two times, and does throw an error when it was shut twice.

To make these two behaviors consistently, this patch implements a strategy,
which will throw an error when init/shutdown are called continuously.

Fix #66