RobotWebTools/rclnodejs #72: Failed to npm install on macOS

qiuzhong, 2017-08-22:
## Description:
I can run the `demo_nodes_cpp` and `demo_nodes_py` with talker and listener successfully on macOS, however I still failed to install rclnodejs with `npm install`.

## Steps:
```
$ source install/local_setup.bash
$ mkdir RobotWebTools && cd RobotWebTools
$ git clone https://github.com/RobotWebTools/rclnodejs.git && cd rclnodejs
$ npm install

> ref@1.3.4 install /Users/zjs/Downloads/ros2_sw/ros2/RobotWebTools/rclnodejs/node_modules/ref
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/binding.o
  SOLINK_MODULE(target) Release/binding.node
clang: warning: libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9 [-Wdeprecated]

> ffi@2.2.0 install /Users/zjs/Downloads/ros2_sw/ros2/RobotWebTools/rclnodejs/node_modules/ffi
> node-gyp rebuild

  CC(target) Release/obj.target/ffi/deps/libffi/src/prep_cif.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/types.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/raw_api.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/java_raw_api.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/closures.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/x86/ffi.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/x86/ffi64.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/x86/darwin.o
  CC(target) Release/obj.target/ffi/deps/libffi/src/x86/darwin64.o
  LIBTOOL-STATIC Release/libffi.a
  CXX(target) Release/obj.target/ffi_bindings/src/ffi.o
../src/ffi.cc:111:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("RTLD_NEXT").ToLocalChecked(), WrapPointer((char *)RTLD_NEXT), static_cast<PropertyAttribute>(ReadOnly | DontDelete));
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:114:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("RTLD_DEFAULT").ToLocalChecked(), WrapPointer((char *)RTLD_DEFAULT), static_cast<PropertyAttribute>(ReadOnly | DontDelete));
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:117:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("RTLD_SELF").ToLocalChecked(), WrapPointer((char *)RTLD_SELF), static_cast<PropertyAttribute>(ReadOnly|DontDelete));
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:120:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("RTLD_MAIN_ONLY").ToLocalChecked(), WrapPointer((char *)RTLD_MAIN_ONLY), static_cast<PropertyAttribute>(ReadOnly|Don...
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:123:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("FFI_ARG_SIZE").ToLocalChecked(), Nan::New<Uint32>((uint32_t)sizeof(ffi_arg)), static_cast<PropertyAttribute>(ReadOn...
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:124:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("FFI_SARG_SIZE").ToLocalChecked(), Nan::New<Uint32>((uint32_t)sizeof(ffi_sarg)), static_cast<PropertyAttribute>(Read...
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:125:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("FFI_TYPE_SIZE").ToLocalChecked(), Nan::New<Uint32>((uint32_t)sizeof(ffi_type)), static_cast<PropertyAttribute>(Read...
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:126:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("FFI_CIF_SIZE").ToLocalChecked(), Nan::New<Uint32>((uint32_t)sizeof(ffi_cif)), static_cast<PropertyAttribute>(ReadOn...
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
../src/ffi.cc:132:11: warning: 'ForceSet' is deprecated [-Wdeprecated-declarations]
  target->ForceSet(Nan::New<String>("HAS_OBJC").ToLocalChecked(), Nan::New<Boolean>(hasObjc), static_cast<PropertyAttribute>(ReadOnly | DontDelete));
          ^
/Users/zjs/.node-gyp/6.10.2/include/node/v8.h:2695:22: note: 'ForceSet' has been explicitly marked deprecated here
                bool ForceSet(Local<Value> key, Local<Value> value,
                     ^
9 warnings generated.
  CXX(target) Release/obj.target/ffi_bindings/src/callback_info.o
  CXX(target) Release/obj.target/ffi_bindings/src/threaded_callback_invokation.o
  SOLINK_MODULE(target) Release/ffi_bindings.node
clang: warning: libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9 [-Wdeprecated]
ld: warning: could not create compact unwind for _ffi_call_unix64: does not use RBP or RSP based frame

> rclnodejs@0.0.1 install /Users/zjs/Downloads/ros2_sw/ros2/RobotWebTools/rclnodejs
> node-gyp rebuild

  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
../src/executor.cpp:17:10: fatal error: 'rcl/error_handling.h' file not found
#include <rcl/error_handling.h>
         ^
1 error generated.
make: *** [Release/obj.target/rclnodejs/src/executor.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/zjs/.nvm/versions/node/v6.10.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Darwin 16.7.0
gyp ERR! command "/Users/zjs/.nvm/versions/node/v6.10.2/bin/node" "/Users/zjs/.nvm/versions/node/v6.10.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/zjs/Downloads/ros2_sw/ros2/RobotWebTools/rclnodejs
gyp ERR! node -v v6.10.2
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 

npm ERR! Darwin 16.7.0
npm ERR! argv "/Users/zjs/.nvm/versions/node/v6.10.2/bin/node" "/Users/zjs/.nvm/versions/node/v6.10.2/bin/npm" "install"
npm ERR! node v6.10.2
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! rclnodejs@0.0.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.0.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the rclnodejs package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs rclnodejs
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls rclnodejs
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/zjs/Downloads/ros2_sw/ros2/RobotWebTools/rclnodejs/npm-debug.log
```

## Environment:
* OS: macOS Sierra Version 10.12.6
* ros2/ros2: [ab20dea](https://github.com/ros2/ros2/commit/ab20dea002c213079774167a77997a55a09ea798)

minggangw, 2017-08-22:
Blocked by #46

minggangw, 2017-09-18:
This should be fixed by commit 0e8301549, also there is a pr #101, which demonstrates how to enable the ci on mac platform by circleci, but the circleci.com doesn't offer a free service on mac, so this pr will be pending. @qiuzhong please verify this issue on your own mac.

minggangw, 2017-09-19:
Move to M3

minggangw, 2017-09-29:
We can fix this issue by downloading the binary ros package from http://ci.ros2.org/view/packaging/job/packaging_osx/, or reference https://circleci.com/gh/RobotWebTools to configure.

closed 2017-09-29 by minggangw

qiuzhong, 2017-10-09:
Actually, we finally got through the compilation on macOS 10.12.6. It turns out that the SIP (System Integrity Protection) should be disabled.
 More details see [this](https://developer.apple.com/library/content/documentation/Security/Conceptual/System_Integrity_Protection_Guide/ConfiguringSystemIntegrityProtection/ConfiguringSystemIntegrityProtection.htm):
```
$ csrutil enable
```

minggangw, 2017-10-09:
That's clear, maybe the default value of SIP is disabled in 10.11, so I haven't met this problem, thanks!