RobotWebTools/rclnodejs #74: Why does node.createTimer parameter period use nanosecond as unit?

qiuzhong, 2017-08-23:
The `node.createTimer(peroid, callback)` method use nanosecond as the unit of peroid parameter. 1 second equals `10 ** 9` nanoseconds. This is a big number and can easily get wrong for users. 

The [rclpy](https://github.com/ros2/rclpy) implementation uses second as the unit and multiply a [constant](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/constants.py#L15) `1000 * 1000 * 1000` before pass the parameters to native RCL API.

Suggest to use second as the parameter unit as well in rclnodejs.

mmccool, 2017-08-23:
This would be fine in theory... as long as the number is interpreted as a float and precision is preserved, so it's still possible to set a timer with nanosecond resolution.

One of my major annoyances with Node.js is the fact that the argument to SetTimeout and SetInterval are interpreted as integers, in units of milliseconds, so it's not possible to set a finer-grained timer (even if the underlying system would support it).  One simple fix to SetInterval would simply be to interpret this number as a floating-point value (which, since JavaScript is untyped anyhow, shouldn't break anything, although it might need an extra function entry point in the underlying library).

Although one could also argue with making the unit ms to be consistent with SetInterval... although still interpreting as a floating-point value for the reasons above.

mmccool, 2017-08-23:
As long as (a) the unit is clear and (b) floating-point values are accepted so there is no problem with quantization (or range...).   As to what default unit is used, it boils down to what people are most likely to expect (ROS users?  Node.js users?)

minggangw, 2017-09-19:
Move to M3

minggangw, 2017-09-22:
I investigated the `rclpy`, `rclcpp` and `rcl`, the result is:

**rclpy** uses para in `second` when creating a [`timer`](https://github.com/ros2/rclpy/blob/0fbf30c4a9ffe9b4edb77c4888d07a61ed921ef9/rclpy/rclpy/node.py#L187)
**rclcpp** uses `ms` when creating a [`timer`](https://github.com/ros2/rclcpp/blob/5a99bbdc6e0294cc307e859be7ca9c7b57444979/rclcpp/include/rclcpp/node.hpp#L219)
**rcl** uses `ns` to pass it to middle ware.

I think we'd better use `ms` as default here, the granularity of millisecond seems more proper here, what's your suggestion?

minggangw, 2017-09-29:
Closed, fixed by b2187ac1a

closed 2017-09-29 by minggangw

qiuzhong, 2017-10-09:
Verified. Now we can use millissecond as the unit of rclnodejs timer.