RobotWebTools/rclnodejs #85: Critical crash when running npm test

qiuzhong, 2017-08-30:
## Description:
I encountered this critical crash when running `npm test`
```
npm test

> rclnodejs@0.0.1 test /home/qiuzhong/ros2_sw/RobotWebTools/rclnodejs
> mocha test/test-*.js

  Node destroy testing
    ✓ node.destroy()
    ✓ node.destroy() twice
    ✓ node.destroy() corrupted
    ✓ node.destory timers
    ✓ node destory entities
    ✓ node currupted node handle

  rclnodejs module existance testing
    rclnodejs module members
      ✓ createNode method should exist
      ✓ init method should exist
      ✓ regenerateAll method should exist
      ✓ require method should exist
      ✓ shutdown method should exist
      ✓ spin method should exist

  rclnodejs class existance testing
    Client class
      ✓ sequenceNumber property should exist
      ✓ sendRequest method should exist
    Node class
      ✓ createClient method should exist
      ✓ createPublisher method should exist
      ✓ createService method should exist
      ✓ createSubscription method should exist
      ✓ createTimer method should exist
      ✓ destroy method should exist
    Publisher class
      ✓ topic property should exist
      ✓ publish method should exist
    Timer class
      ✓ period property should exist
      ✓ cancel method should exist
      ✓ isCanceled method should exist
      ✓ isReady method should exist
      ✓ reset method should exist
      ✓ timeSinceLastCall method should exist
      ✓ timeUntilNextCall method should exist

  Node destroy testing
    ✓ rclnodejs.init()
    ✓ rclnodejs.init() & rclnodejs.shutdown()
    ✓ rclnodejs init shutdown sequence
    ✓ rclnodejs double init
    ✓ rclnodejs double shutdown
    ✓ rclnodejs create node without init

  rclnodejs interactive testing
    Publisher/Subscription
      ✓ Publisher/Subscription (124ms)
    Client/Service
      ✓ Client/Service (317ms)

  rclnodejs node test suite
    createNode method testing
      ✓ Try creating a node
      ✓ Try creating a node with an invalid name
      ✓ Try creating a node with an invliad relative nampespace
      ✓ Try creating a node with an invalid absolute namespace
      ✓ Try creating a node with invalid type of parameters

[... 235 lines cut ...]
7f7fc60ae000-7f7fc60af000 rw-p 00007000 08:01 30939024                   /lib/x86_64-linux-gnu/librt-2.23.so
7f7fc60af000-7f7fc60b2000 r-xp 00000000 08:01 30939027                   /lib/x86_64-linux-gnu/libdl-2.23.so
7f7fc60b2000-7f7fc62b1000 ---p 00003000 08:01 30939027                   /lib/x86_64-linux-gnu/libdl-2.23.so
7f7fc62b1000-7f7fc62b2000 r--p 00002000 08:01 30939027                   /lib/x86_64-linux-gnu/libdl-2.23.so
7f7fc62b2000-7f7fc62b3000 rw-p 00003000 08:01 30939027                   /lib/x86_64-linux-gnu/libdl-2.23.so
7f7fc62b3000-7f7fc62d9000 r-xp 00000000 08:01 30939016                   /lib/x86_64-linux-gnu/ld-2.23.so
7f7fc64b2000-7f7fc64b8000 rw-p 00000000 00:00 0 
7f7fc64d1000-7f7fc64d2000 rw-p 00000000 00:00 0 
7f7fc64d2000-7f7fc64d3000 ---p 00000000 00:00 0 
7f7fc64d3000-7f7fc64d8000 rw-p 00000000 00:00 0 
7f7fc64d8000-7f7fc64d9000 r--p 00025000 08:01 30939016                   /lib/x86_64-linux-gnu/ld-2.23.so
7f7fc64d9000-7f7fc64da000 rw-p 00026000 08:01 30939016                   /lib/x86_64-linux-gnu/ld-2.23.so
7f7fc64da000-7f7fc64db000 rw-p 00000000 00:00 0 
7ffcb114d000-7ffcb1170000 rw-p 00000000 00:00 0                          [stack]
7ffcb11f4000-7ffcb11f6000 r--p 00000000 00:00 0                          [vvar]
7ffcb11f6000-7ffcb11f8000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)
npm ERR! Test failed.  See above for more details.

```

## Commit Id
* rclnodejs: [85c0185](https://github.com/RobotWebTools/rclnodejs/commit/85c0185f0282272b1f34bed774f8e489d2ac1e77)
* ros2.repos md5: 84531206966a83c0743af50cb7026a7e

minggangw, 2017-08-30:
As this issue was caused by the wrong environment configuration, and can't reproduce on the latest `master` branch of `ROS2.0` with `develop` branch of `rclnodejs`, so I think we can close it now.@qiuzhong

qiuzhong, 2017-08-31:
Agree with you. Let's close it.

closed 2017-08-31 by qiuzhong