RobotWebTools/rclnodejs #87: QoS members are undefined in constructor

qiuzhong, 2017-08-30:
In the class `QoS` constructor, the member is initialized as `undefined` like this:
```
class QoS {
  constructor() {
    this._history = undefined;
    this._depth = undefined;
    this._reliability = undefined;
    this._durability = undefined;
    this._avoidRosNameSpaceConventions = undefined;
  }
...
```

IMHO, this is not a good initialization. After you new a QoS object, call the getter of any property you will get an `undefined` type.
However, each getter has a special type for the returned value in the docs.
I would suggest initialize them as all `false` evaluation, that is:
* For boolean getter, initialize it as `false`
* For string getter, initialize it as `''`
* For number getter, initialize it as `0`
* For object getter, initialize it as `null`

minggangw, 2017-08-30:
Good suggestion, I am gonna take this, thanks!

closed 2017-09-04 by minggangw

qiuzhong, 2017-09-04:
Verified. The QoS object properties are initialized in a good way and not undefined now.

PR #89 (2017-09-04): [QoS]Set default value in QoS constructor
Currently, the properties of QoS object will be initialized as undefined,
which is not a good practice.

This patch will set the default value for each property in the
constructor, and update the corresponding document. Besides, we remove
some unnecessary dependencies in the package.json.

Fix #87
commit e8ffb5db (2017-09-04): [QoS]Set default value in QoS constructor

Currently, the properties of QoS object will be initialized as undefined,
which is not a good practice.

This patch will set the default value for each property in the
constructor, and update the corresponding document. Besides, we remove
some unnecessary dependencies in the package.json.

Fix #87