RobotWebTools/rclnodejs #106: No boundary checking for message type ColorRGBA properties

qiuzhong, 2017-09-19:
Currently, the `ColorRGBA` type of package `std_msgs` has 4 properties: `r, g, b, a`. Regularly, `r, g, b` should be integers ([0, 255]) and `a` should be a float([0.0, 1.0]). But in the rclnodejs implementation, they're all `float32`. Meanwhile, there is no boundary checking for the properties. So this code
```
rclnodejs.init().then(() => {
  var ColorRGBA = rclnodejs.require('std_msgs').msg.ColorRGBA;
  var msg = new ColorRGBA();
  msg.a = 1.1;
  msg.r = 256;
}).catch((err) => {
  console.log(err);
});
```
doesn't throw any errors.

minggangw, 2017-10-13:
Hi @qiuzhong would you please verify whether the `rclpy` has the boundary checking on ColorRGBA. If user add a new kind of `.msg` not **ColorRGBA**, I suppose this checking should be done by the caller himself.

qiuzhong, 2017-10-16:
@minggangw , I checked the `ColorRGBA` type for the `rclpy` and found it only requires the `r, g, b, a` be `float` type, no boundary checking either.

minggangw, 2017-10-16:
So we suggest the owner, who use the `.msg` file, to validate the data transferred, and `.msg` file supports to define constant variables, e.g. [this](https://github.com/ros2/common_interfaces/blob/6a78a1d6aceadccc5871311cc49156ef952bb3aa/shape_msgs/msg/SolidPrimitive.msg#L4), which can remind of checking the value if it's necessory. So shall we close this issue @qiuzhong ?

qiuzhong, 2017-10-16:
Agreed to close this issue. Let me close it.

closed 2017-10-16 by qiuzhong