RobotWebTools/rclnodejs #110: *ArrayWrapper cannot be used to createPublisher

qiuzhong, 2017-09-20:
Take the `Byte` as example, there is a `ByteArrayWrapper` class in the generated `generated/std_msgs/std_msgs__msg__Byte.js`. It stands for an array of the `Byte` array message type. However, it cannot be used to `createPublisher`. This code
```
const rclnodejs = require('./rclnodejs/index.js');

rclnodejs.init().then(() => {
  var node = rclnodejs.createNode('byte_publisher');
  var Byte = rclnodejs.require('std_msgs').msg.Byte;
  var ByteArray = Byte.ArrayType;
  var msg = new ByteArray(3);
  msg.setWrappers([1, 2, 3]);
  // console.log(`${msg.data}`);
  console.log(msg.data[0].data); // print 1
  var publisher = node.createPublisher(ByteArray, 'chatter');
  setInterval(() => {
    console.log(`Publishing: ${msg.data}`);
    publisher.publish(msg);
  }, 500);
  rclnodejs.spin(node);
}).catch((err) => {
  console.log(err);
});

```
will lead a `TypeError` when calls `createPublisher`:
```
1
TypeError: typeClass.type is not a function
    at Function.createPublisher (/home/qiuzhong/Workspace/Work/ros2/tmp/rclnodejs/lib/publisher.js:57:26)
    at Node.createPublisher (/home/qiuzhong/Workspace/Work/ros2/tmp/rclnodejs/lib/node.js:111:31)
    at rclnodejs.init.then (/home/qiuzhong/Workspace/Work/ros2/tmp/publisher.js:106:24)
    at process._tickCallback (internal/process/next_tick.js:109:7)
    at Module.runMain (module.js:606:11)
    at run (bootstrap_node.js:389:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:504:3

```

minggangw, 2017-09-22:
After I was investigating this issue, I found there is a misunderstanding of the usage of `Array` type. The fact is that you can't simply create an **Array** of some kind of Message directly, like above. Instead, you have to create a new type of `.msg` file, 

e.g. If you want to use the array of `string`, please create a new file named `Names.msg` and write below and save.
```
string[] names
```
Then leverage  `rosidl_generator_c` and `rosidl_typesupport_introspection_c` to generate corresponding c libraries, which will be used by `rclnodejs`. This is a little complicated, actually you can just put this msg file under the `common_interfaces/std_msgs/msg/` of your ros2 source folder, and add a new line  `"msg/Names.msg"` to the `CMakeLists.txt`, then build the ros again. All you need will be gererated.

Finally, test like:
```javascript
const rclnodejs = require('./rclnodejs/index.js');

rclnodejs.init().then(() => {
  var node = rclnodejs.createNode('byte_publisher');
  var Names = rclnodejs.require('std_msg').msg.Names; // Here we assume you put the new msg file into the std_msg
  var msg = new Names;
  msg.names = ['Tom', 'Jerry'];
  var publisher = node.createPublisher(Names, 'chatter');
  setInterval(() => {
    publisher.publish(msg);
  }, 500);
  rclnodejs.spin(node);
}).catch((err) => {
  console.log(err);
});
```
The rule is that you can't use the **Array** explicitly.

qiuzhong, 2017-09-28:
@minggangw , I tried this. The `generated/std_msgs/std_msgs__msg__Names.js` was generated. However, I got crash when I ran the the code again:
```
<--- Last few GCs --->

   10994 ms: Mark-sweep 1382.1 (1434.2) -> 1382.1 (1434.2) MB, 1159.2 / 0.0 ms [allocation failure] [GC in old space requested].
   12148 ms: Mark-sweep 1382.1 (1434.2) -> 1382.1 (1417.2) MB, 1153.5 / 0.0 ms [last resort gc].
   13306 ms: Mark-sweep 1382.1 (1417.2) -> 1382.1 (1417.2) MB, 1158.5 / 0.0 ms [last resort gc].

<--- JS stacktrace --->

==== JS stack trace =========================================

Security context: 0x1af3d33cfb39 <JS Object>
    1: setup [/home/qiuzhong/Workspace/Work/rclnodejs/ros2/RobotWebTools/rclnodejs/node_modules/array-index/index.js:~169] [pc=0x171f4bc4e4d5] (this=0x3be6fd9c6c9 <JS Global Object>,index=5424016)
    2: ensureLength [/home/qiuzhong/Workspace/Work/rclnodejs/ros2/RobotWebTools/rclnodejs/node_modules/array-index/index.js:~139] [pc=0x171f4bc5edd4] (this=0x3be6fd9c6c9 <JS Global Object>,self=0xa040...

FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory
 1: node::Abort() [node]
 2: 0x109bf8c [node]
 3: v8::Utils::ReportApiFailure(char const*, char const*) [node]
 4: v8::internal::V8::FatalProcessOutOfMemory(char const*, bool) [node]
 5: v8::internal::Factory::NewFillerObject(int, bool, v8::internal::AllocationSpace) [node]
 6: v8::internal::Runtime_AllocateInTargetSpace(int, v8::internal::Object**, v8::internal::Isolate*) [node]
 7: 0x171f4b9079a7
Aborted (core dumped)
```

minggangw, 2017-09-28:
I am going to re-check this problem, thanks!

qiuzhong, 2017-09-28:
To show the message sent, I added a line `console.log(msg.names);` before calling the `publish`. However, if I remove this line, then it seems not crash. The subscription node cannot receive any message in both cases.

qiuzhong, 2017-10-10:
The subscription can receive `Names` type message from publisher now. But the crash remains if `msg.names` is referred before `publish` method calling.

closed 2017-10-11 by minggangw

qiuzhong, 2017-10-11:
Verified. #137 fixes the crash.

PR #137 (2017-10-11): [rosidl_gen]Get the property of array type from the wrapper object
Currently when getting the property of array type, we will get the value
from ref-struct object directly, which has not been assigned.

To fix this, we must get/set the property through the wrapper
consistently.

Fix #110
commit 3249daa2 (2017-10-11): [rosidl_gen]Get the property of array type from the wrapper object

Currently when getting the property of array type, we will get the value
from ref-struct object directly, which has not been assigned.

To fix this, we must get/set the property through the wrapper
consistently.

Fix #110