RobotWebTools/rclnodejs #111: More detailed documentation for Windows build

qiuzhong, 2017-09-21:
Currently, the `rclnodejs` supported Windows 8.1 & 10, however, users may encounter quite annoying issues during building and testing.

* `npm run lint`: this uses `wget` utility, which is not installed by `choco` during in building ros2. 
* Even users install `wget` via `choco install -y wget`, there is still `cpplint` issue when running `npm run lint`:
  ```
  Downloading the cpplint...
  Running the cpplint...

  C:\dev\ros2\wangminggang\rclnodejs\scripts\cpplint.js:35
          throw Error('cpplint failed.');
          ^

  Error: cpplint failed.
      at Error (native)
      at exec (C:\dev\ros2\wangminggang\rclnodejs\scripts\cpplint.js:35:15)
      at ChildProcess.exithandler (child_process.js:205:5)
      at emitTwo (events.js:106:13)
      at ChildProcess.emit (events.js:191:7)
      at maybeClose (internal/child_process.js:920:16)
      at Process.ChildProcess._handle.onexit (internal/child_process.js:230:5)
  ```

* rclnodejs depends on thirdparty libraries, so before building rclnodejs, we should run `git submodule update --init --recursive`.

minggangw, 2017-09-22:
Reopen, as we have more work to do here.

minggangw, 2017-09-29:
As 49a101e6b560 has been merged, this issue can be closed now.

closed 2017-09-29 by minggangw

qiuzhong, 2017-10-09:
The CI can get through this issue. But it still lacks `wget` dependency installation in the README in case the users tried to build on Windows manually.

PR #114 (2017-09-22): [README]Update the content
As we have enabled all platforms, including Linux/MacOS/Windows, the
README needs to be updated to align with the new progress:

1. Update the link of travis-ci.
2. Add the step of getting the code.
3. Delete the linux only support description.

Fix #111
commit 663b4697 (2017-09-22): [README]Update the content

As we have enabled all platforms, including Linux/MacOS/Windows, the
README needs to be updated to align with the new progress:

1. Update the link of travis-ci.
2. Add the step of getting the code.
3. Delete the linux only support description.

Fix #111