RobotWebTools/rclnodejs #123: [UT]Make the test cases pass for all platforms

qiuzhong, 2017-09-28:
Currently, all the test cases are developed on Ubuntu Linux 16.04 LTS and passed when submitting. However, there are platform differences for ROS2 or other dependencies. For example, on Linux and macOS, Python 2.x and Python3 can coexist (Python 2.x -> python, Python 3.x -> python3). On Windows, all version Python executables are `python.exe`.  Some test cases may fail on Windows due to the lack of platform difference considerations.

`npm test` should pass on all the platforms(Windows/macOS/Linux), so all the platform differences should be considered.

minggangw, 2017-09-29:
Since at current stage, some cases don't pass on platform, like Windows or MacOS, we'd better have a blacklist to disable them temporarily and let the rest of cases check at least.

minggangw, 2017-10-12:
Status update of unit test:
- **Linux** : 100% passed
- **MacOS**:  [1 case](https://github.com/RobotWebTools/rclnodejs/blob/develop/test/blacklist.json#L3) is disabled (issue #127), other passed. 
- **Window**:  [5 cases](https://github.com/RobotWebTools/rclnodejs/blob/develop/test/blacklist.json#L4) are disable, other passed. During running the disabled ones, `mocha` will terminate without any errors.

minggangw, 2017-10-19:
Call stack when crashing on macOS, seems we freed something that should not be
```
node(74099,0x7fffa3985340) malloc: *** error for object 0x103000130: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Abort trap: 6 (core dumped)
```

```
* thread #1, stop reason = signal SIGSTOP
  * frame #0: 0x00007fff6a40afce libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff6a548150 libsystem_pthread.dylib`pthread_kill + 333
    frame #2: 0x00007fff6a36732a libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff6a464866 libsystem_malloc.dylib`free + 521
    frame #4: 0x00000001004af5d0 node`v8::internal::ArrayBufferTracker::FreeDead(bool) + 112
    frame #5: 0x00000001004bc03e node`v8::internal::Heap::Scavenge() + 2910
    frame #6: 0x00000001004ba16a node`v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::GCCallbackFlags) + 1322
    frame #7: 0x00000001004b974d node`v8::internal::Heap::CollectGarbage(v8::internal::GarbageCollector, char const*, char const*, v8::GCCallbackFlags) + 701
    frame #8: 0x000000010047a2cb node`v8::internal::Factory::NewFillerObject(int, bool, v8::internal::AllocationSpace) + 107
    frame #9: 0x0000000100705fc7 node`v8::internal::Runtime_AllocateInTargetSpace(int, v8::internal::Object**, v8::internal::Isolate*) + 503
    frame #10: 0x00003970707079a7
```

closed 2017-10-30 by minggangw

qiuzhong, 2017-10-30:
Verified. Just tested on Windows 10 & macOS 10.12.6 and all tests passed!

PR #158 (2017-10-30): [UT]Enable all tests on Linux/macOS/Windows
Fix #123
commit cd141396 (2017-10-30): [UT]Enable all tests on Linux/macOS/Windows

Fix #123