RobotWebTools/rclnodejs #134: Failed to receive complex object type message cross the language nodes

qiuzhong, 2017-10-09:
For the `JointState` complex object type messages, in a **Python publisher - rclnodejs subscription** environment, the subscription cannot receive the Python publisher:
**In terminal 1, `python3 py_publisher.py`**
```
import sys
import rclpy
from time import sleep
from std_msgs.msg import *
from builtin_interfaces.msg import *
from sensor_msgs.msg import *

def main():
  rclpy.init()
  node = rclpy.create_node('py_jointstate_publisher')
  publisher = node.create_publisher(JointState, 'JointState_channel')
  time = Time()
  time.sec = 123456
  time.nanosec = 789
  header = Header()
  header.stamp = time
  header.frame_id = 'main frame'
  msg = JointState()
  msg.header = header
  msg.name = ['Tom', 'Jerry']
  msg.position = [1.0, 2.0]
  msg.velocity = [2.0, 3.0]
  msg.effort = [4.0, 5.0, 6.0]

  while True:
    print(msg.name)
    print(msg.position)
    publisher.publish(msg)
    sleep(0.5)

if __name__ == '__main__':
  main()
```

**In terminal 2, node subscription.js**:
```
const rclnodejs = require('./rclnodejs/index.js');

rclnodejs.init().then(() => {
  var node = rclnodejs.createNode('jointstate_js_publisher');
  const Timer = rclnodejs.require('builtin_interfaces').msg.Time;
  const Header = rclnodejs.require('std_msgs').msg.Header;
  const JointState = rclnodejs.require('sensor_msgs').msg.JointState;
  var subscription = node.createPublisher(JointState, 'JointState_channel', (msg) => {
    console.log('received');
    console.log(msg.name);
  });
  rclnodejs.spin(node);
}).catch((err) => {
  console.log(err);
});

```
The rclnodejs subscription hanged and didn't receive any message from Python publisher. Python publisher - Python subscription works fine. If the message type is `ColorRGBA` or `Header`, the example will also work fine.

qiuzhong, 2017-10-09:
@minggangw , I doubt it's related to the array message type feature.

minggangw, 2017-10-09:
Do you want to design a case that using a python publisher and a rclnodejs subscription to test the compatibility (Python<=>Node.js)? If that' your aim, I noticed that you created a publisher instead of a subscription in the JavaScript code 
```
var subscription = node.createPublisher(JointState, 'JointState_channel', (msg) => {
```
Is it a typo?

P.S. I also tried to create a subscription, it seems that everything works fine :)

qiuzhong, 2017-10-09:
Oh, you're right! I used the wrong create function. By replace `createSubscription` the rclnodejs node can receive the message.

closed 2017-10-09 by qiuzhong