RobotWebTools/rclnodejs #136: rclnodejs.init() and npm run lint require different Python versions

qiuzhong, 2017-10-10:
Python 2 and Python 3 can coexist perfectly on macOS and Linux by two different python executables (python and python3). However, this is not true on Windows. On Windows, there is only `python.exe` for both versions. 

`rclnodejs.init()` requires Python 3 while `npm run lint` requires Python 2. This is not a problem for the CI on Windows as **appveyor.yml** change the `PATH` environment variable before running tests (will call the init method). However, it's very annoy for Windows users, especially running `npm run lint` and `npm test`.

If Python 3 is first searched in `PATH`, then:
```
>npm run lint

> rclnodejs@0.0.1 lint C:\dev\ros2\RobotWebTools\rclnodejs
> eslint --max-warnings=0 index.js scripts lib example rosidl_gen rosidl_parser test && node ./scripts/cpplint.js

Downloading the cpplint...
Running the cpplint...

C:\dev\ros2\RobotWebTools\rclnodejs\scripts\cpplint.js:35
        throw Error('cpplint failed.');
        ^

Error: cpplint failed.
    at Error (native)
    at exec (C:\dev\ros2\RobotWebTools\rclnodejs\scripts\cpplint.js:35:15)
    at ChildProcess.exithandler (child_process.js:205:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:920:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:230:5)
```

If Python 2 is first searched in `PATH`, some rosidl will not be generated.
```
Error: ENOENT: no such file or directory, scandir 'C:\dev\ros2\RobotWebTools\rclnodejs\generated\std_msgs'
      at Error (native)
      at Object.fs.readdirSync (fs.js:953:18)
      at Object.loadInterfaceInfos (lib\interface_loader.js:27:20)
      at Object.require (index.js:127:33)
      at Context.<anonymous> (test\test-rosidl-message-generator.js:68:31)
```

We should at least clarify the requirements in README to help Windows users get rid of the errors above.

qiuzhong, 2017-10-10:
@minggangw

minggangw, 2017-10-10:
Absolutely, we should make user be aware of this.

qiuzhong, 2017-10-10:
My suggestion is that we should keep Python 3 be searched first in `PATH` and rename Python 2 executable as `pyhon2.exe` and then use `python2` at line 31 [cpplint.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/scripts/cpplint.js):
> https://github.com/RobotWebTools/rclnodejs/blob/develop/scripts/cpplint.js

minggangw, 2017-10-10:
How about using [python launcher for windows](https://docs.python.org/3.3/using/windows.html#python-launcher-for-windows)

qiuzhong, 2017-10-10:
I have thought again about my suggestion and it's not a good solution.
Python launcher for windows is a much better one!

qiuzhong, 2017-10-13:
All the test cases related to the Python process launch is used the Python launched solution on Windows in PR #140

minggangw, 2017-10-13:
Great! I think we can try to enable more tests on MacOS & Windows to see if there is still failure happens later.

PR #139 (2017-10-13): Leverage python launcher on windows platform
Fix #136
commit 21ffbbd7 (2017-10-13): Leverage python launcher on windows platform

Fix #136
closed 2017-10-13 by minggangw