RobotWebTools/rclnodejs #141: npm install fails

elbrini, 2017-10-16:
Hi, 
I am trying to use rclnodejs with ROS2. I followed the instructions from the README but I am encountering an issue while building rclnodejs package with npm install.
Below it the error trace:
```
> rclnodejs@0.0.1 install /home/scalaman/Freelance/direct/realbotics/rclnodejs
> node-gyp rebuild

gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make : on entre dans le répertoire « /home/scalaman/Freelance/direct/realbotics/rclnodejs/build »
  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
../src/executor.cpp:17:32: fatal error: rcl/error_handling.h: Aucun fichier ou dossier de ce type
compilation terminated.
rclnodejs.target.mk:114 : la recette pour la cible « Release/obj.target/rclnodejs/src/executor.o » a échouée
make: *** [Release/obj.target/rclnodejs/src/executor.o] Erreur 1
make : on quitte le répertoire « /home/scalaman/Freelance/direct/realbotics/rclnodejs/build »
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/scalaman/.nvm/versions/node/v6.11.4/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Linux 4.4.0-34-generic
gyp ERR! command "/home/scalaman/.nvm/versions/node/v6.11.4/bin/node" "/home/scalaman/.nvm/versions/node/v6.11.4/lib/node_modules/npm/node_modules/node
-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/scalaman/Freelance/direct/realbotics/rclnodejs
gyp ERR! node -v v6.11.4
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 

npm ERR! Linux 4.4.0-34-generic
npm ERR! argv "/home/scalaman/.nvm/versions/node/v6.11.4/bin/node" "/home/scalaman/.nvm/versions/node/v6.11.4/bin/npm" "install"
npm ERR! node v6.11.4
....
```

qiuzhong, 2017-10-17:
Thanks to Google translation, the error log specified that the `rcl/error_handling.h` was not found during installing the `rclnodejs`. This header file is from ROS2.
Please make sure you have compiled the ROS2 and do the following action:
```
source /install/local_setup.bash
```

Then you can go to `rclnodejs` directory and run `npm install`

elbrini, 2017-10-17:
Oh, I am sorry I didn't pay attention that the errors are in french.
Thanks for the reply, it fixed my issue :)

closed 2017-10-17 by elbrini