RobotWebTools/rclnodejs #144: Got warnings when running npm test

qiuzhong, 2017-10-18:
In PR #142 , I added a test case to test if a rclnodejs client and rclpy service can work together. The rclnodejs client sends a request to add 1 and 2, the rclpy service receives the request and performs the addition and send a response.

Here is my test code:
```
  describe('Node.js client', function() {
    it('Node.js client should work with Python service', function(done) {
      var node = rclnodejs.createNode('js_add_client');
      const AddTwoInts = rclnodejs.require('example_interfaces').srv.AddTwoInts;
      var destroy = false;

      var pyService = utils.launchPythonProcess([`${__dirname}/py/service.py`]);
      var client = node.createClient(AddTwoInts, 'py_js_add_service');
      let request = new AddTwoInts.Request();
      request.a = 1;
      request.b = 2;

      var timer = setInterval(() => {
        client.sendRequest(request, (response) => {
          if (!destroy) {
            assert.deepStrictEqual(response.sum, 3);
            clearInterval(timer);
            node.destroy();
            pyService.kill('SIGINT');
            destroy = true;
            done();
          }      
        });
      }, 100);

      rclnodejs.spin(node);
    });
  });
```

This case can pass on Linux, how ever I got the warnings when running `npm test`:
```
  Node destroy testing
    ✓ rclnodejs.init()
    ✓ rclnodejs.init() & rclnodejs.shutdown()
    ✓ rclnodejs init shutdown sequence
    ✓ rclnodejs double init
    ✓ rclnodejs double shutdown

>>> [rcutils|error_handling.c:148] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/client.c:279'

with this new error message:

  'rcl_init() has not been called, at /home/qiuzhong/Workspace/Work/ros2/src/ros2/rcl/rcl/src/rcl/node.c:121'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
    ✓ rclnodejs create node without init

```

The warnings occured when mocha was executing [test-init-shutdown.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/test/test-node.js). However, if I run test-init-shutdown.js individually, no such warnings.

qiuzhong, 2017-10-18:
I have no idea why this happened. @minggangw , would you please take a look and give me some clues?

minggangw, 2017-10-18:
Through the log, there are two errors. One is generated by new test case you added (client<=>service), the other is generated by [create node without init](https://github.com/RobotWebTools/rclnodejs/blob/develop/test/test-init-shutdown.js#L92). I think the error caused by the first case lead to the second one. When you were running your new added test case, is there any error?

qiuzhong, 2017-10-18:
There is no error when running the new added test case. 

In terms of the ** create node without init**, it's on purpose in the `assert.throw` statement in test case. Before this PR, it has never complains such warnings. 

And I also found this strange behavior:
* `mocha test-init-shutdown.js test-cross-lang.js`: all pass without any warning
* `mocha test-cross-lang.js test-init-shutdown.js`: all pass with warnings.

qiuzhong, 2017-10-18:
You can check the [CI](https://travis-ci.org/RobotWebTools/rclnodejs/builds/289368947?utm_source=github_status&utm_medium=notification), though the test cases pass and no PR error, there are these warnings in the log.

minggangw, 2017-10-18:
I checked the implementation of `rcl`, the first error occurred at [client.c](https://github.com/ros2/rcl/blob/master/rcl/src/rcl/client.c#L279), which should be reset after it happen. Thus when we were running the `test-init-shutdown.js` later, it will generate an error on purpose, which overwrites the previous one, so the `warning` is telling you that `This error state is being overwritten`.

We can fix this by resetting the error when we failed to take the response in `rclnodejs`, a patch will be followed, thanks!

closed 2017-10-20 by minggangw

qiuzhong, 2017-10-20:
Verifed in PR#151. Fixed

PR #151 (2017-10-20): Reset the error status
We must reset the error status to avoid being overwritten by the next one.
Calling rcl_reset_error() will meet this when an error happens during
taking an response from service.

Fix #144
commit b660215b (2017-10-20): Reset the error status

We must reset the error status to avoid being overwritten by the next one.
Calling rcl_reset_error() will meet this when an error happens during
taking an response from service.

Fix #144