RobotWebTools/rclnodejs #146: Should rclnodejs initialized checking done in Node.js layer?

qiuzhong, 2017-10-19:
Well, in the test case [test-init-shutdown.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/index.js) it `createNode` without even the `rclnode.init()` is called on purpose. Then it will obviously throw an Error.
```
createNode(nodeName, namespace = '') {
    if (typeof (nodeName) !== 'string' || typeof (namespace) !== 'string') {
      throw new TypeError('Invalid argument.');
    }

    let handle = rclnodejs.createNode(nodeName, namespace);
```

My question is I found there is no rclnodejs initialized checking before it transfers the creating node to native code layer. Should it be done in Node.js layer? If so then there will no error in rcl native code and thus no need to reset error at all.

qiuzhong, 2017-10-19:
@minggangw , what do you think?

I just tried to done this in Node.js layer locally. The issue described in #144 still exists. However, it happened in test suite `test-node.js` as creating node with invalid names or namespaces.

minggangw, 2017-10-19:
I think either is OK here, but #144 is another issue, which should be fixed to prevent the warning from `rcl` library.

qiuzhong, 2017-10-20:
Sine it doesn't matter where to check if rcl is initialized, let's close this?

minggangw, 2017-10-20:
OK, let's close this.

closed 2017-10-20 by minggangw