RobotWebTools/rclnodejs #148: node namespace not starts with '/'

qiuzhong, 2017-10-20:
Well, I convert some [rclpy](https://github.com/ros2/rclpy/blob/master/rclpy) [test cases](https://github.com/ros2/rclpy/blob/master/rclpy/test/test_create_node.py#L44) to `rclnodejs` version. However, PR #147 was not working with this.

In rclpy:
```
    def test_create_node_with_empty_namespace(self):
        node_name = 'create_node_test'
        namespace = ''
        node = rclpy.create_node(node_name, namespace=namespace)
        self.assertEqual('/', node.get_namespace())

    def test_create_node_with_relative_namespace(self):
        node_name = 'create_node_test'
        namespace = 'ns'
        node = rclpy.create_node(node_name, namespace=namespace)
        self.assertEqual('/ns', node.get_namespace())
```
That means:
* If a namespace is empty, `get_namespace()` should return '/'; 
* If a namespace doesn't start with '/', `get_namespace()` should add '/' to the head of the namespace;

Should we keep alignment with `rclpy`?

qiuzhong, 2017-10-20:
@minggangw

minggangw, 2017-10-20:
Yes, we will follow the `rclpy`, which also follows the rules of ROS2.0, I will fix this.

qiuzhong, 2017-10-23:
Verified. Fixed by PR #153

PR #153 (2017-10-23): Add '/' when the namespace doesn't stat with it.
Fix #148
commit fa6f1a65 (2017-10-23): Add '/' when the namespace doesn't stat with it.

Fix #148
closed 2017-10-23 by qiuzhong