RobotWebTools/rclnodejs #173: Add istanbul package to the devDependencies

qiuzhong, 2017-11-10:
Well, for the CI, `npm install -g istanbul` is executed before checking code coverage.  So this is not a problem for CI. 

But for end-users, `istanbul` is NOT in the devDependencies field in package.json, so `npm install` will not install it and I got
```
>npm run test

> rclnodejs@0.0.2 test C:\dev\ros2_sw\ros2\RobotWebTools\rclnodejs
> node ./scripts/compile_cpp.js && istanbul cover ./scripts/run_test.js --report lcovonly

'istanbul' is not recognized as an internal or external command,
operable program or batch file.
```

qiuzhong, 2017-11-10:
@minggangw

minggangw, 2017-11-10:
I have submitted pr #172 to remove the execution of istanbul from the `npm test`, which makes the behaviour reasonable as we only need to run this coverage test on ci. So please verify when the pr is merged, thanks!

qiuzhong, 2017-11-10:
Since npm script `coverage` is removed from package.json, close it.

closed 2017-11-10 by qiuzhong