RobotWebTools/rclnodejs #175: Significant memory usage increasing for lots of messages or long-time subscription

qiuzhong, 2017-11-14:
For publishers, neither lots of messages nor long time publishing will raise significant memory usage increase.

For subscriptions, lots of messages or long time publishing will raise a significant memory usage increase.
See the following tables:

| messages | time | publisher memory rising | subscription memory rising |
| --- | --- | --- | --- |
| 100k | | 0.43% | 6.5% |
|  |  24 hours | 0.15% | 15.51% |

Not sure if there is memory leak for the subscription.

minggangw, 2017-11-16:
It seems that the rising of memory usage is irrelevant to the count of messages the subscription has received, but proportional to the duration of running.

qiuzhong, 2017-11-17:
| scenario | time | messags | interval | publisher memory rising | subscription memory rising |
| --- | --- | --- | --- | --- | --- |
| rclpy publisher <-> rclpy subscription| about 1000s | 100k | 10ms | 1.10% | 15.60% |
| rclnodejs publisher <-> rclpy subscription | 16 hours | 57592 | 1000ms | - | 14.66% |

minggangw, 2017-11-17:
From the statistic, the rclpy client as a subscription has an approximate increasement compared with rclnodejs. I am going to retest the same use case. If I get the same result, the memory consumption may be caused by the middle ware of ROS.

Christywl, 2017-12-08:
Using the latest ROS2 and rclnodejs, I also find this issue for publisher/subscription and client/service:

| Scenario | Message | publisher memory rising | subscription memory rising |
| --- | --- | --- | --- |
| 24-hours-publisher.js/24-hours-subscription.js | 100k | 9.37% | 17.09% |

| Scenario | Client Increasing Rate | Service Increasing Rate |
| --- | --- | --- |
| 24-hours-client.js/24-hours-service.js | 0.24% | **24.95%** |

minggangw, 2017-12-11:
Did you test this case on [Ardent Apalone](https://github.com/ros2/ros2/wiki/Release-Ardent-Apalone)?

Christywl, 2017-12-11:
Before that, I run the tests based on https://github.com/ros2/ros2/commit/b12ebdc199c5c41f5ddfda9dabf0abc21a936d6b

hanyia, 2018-05-10:
scenario | execution time | messags | interval | publisher memory rising | subscription memory rising
-- | -- | -- | -- | -- | --
subscription-endurance-test.js/publisher-endurnance-test.js | 19 hours | 3600000K | 10ms | 45.11% | 31.32%

hanyia, 2018-05-11:
Using the latest ROS2 and rclnodejs for PnP test. The detail nodejs result as below:
scenario | execution time | message | interval | publisher memory rising | subscription memory rising
-- | -- | -- | -- | -- | --
subscription-endurance-test.js/publisher-endurnance-test.js | 19 hours | 3600000k | 10ms | 45.11% | 31.32%

qiuzhong, 2018-07-24:
@minggangw , surely I'll make a test and report it based on the new fix.

minggangw, 2018-07-24:
Thanks a lot!

minggangw, 2018-07-26:
Hi @qiuzhong any updates?

qiuzhong, 2018-07-27:
@minggangw , tried with message `std_msgs/msg/Header`. Here is the report:

| messages | time | publisher memory rising | subscription memory rising |
| --- | --- | --- | --- |
| 10k | about 100s | 0.11% | 0.09% |
| 100k | about 1000s | 1.35% | 1.08% |
| 86389 (1s interval) |  24 hours | 0.11% | 0.41% |

From this report, we can see the memory consumption has reduced very significantly.

minggangw, 2018-07-27:
Great, thanks for your effort @qiuzhong, I think we can close this issue now due to #385. Please reopen it if the problem can reproduce in the future.

closed 2018-07-27 by minggangw