RobotWebTools/rclnodejs #176: Sometimes subscriptions lost messages from multiple publishers

qiuzhong, 2017-11-14:
When there are multiple publisher nodes publishing messages simultaneously, sometimes the subscription seems to be blocking.

We need to investigate why some messages are lost. I'll provide more useful information with `rclpy` client.

qiuzhong, 2017-11-20:
I did the same tests for `rclpy` version. The message losing also exists on for the `rclpy` version and the losing rate is ever higher.

minggangw, 2017-11-20:
I suggest keeping an eye on this problem until the fist non-beta release of ROS arrives. If this phenomenon appears constantly then, I think we can submit an issue either for `rclpy` or `rcl`.

Christywl, 2017-12-08:
I see the similar issue during the Client/Server testing. 

multi-clients<-->one-service:

[image]

minggangw, 2017-12-11:
Did you test this case on [Ardent Apalone](https://github.com/ros2/ros2/wiki/Release-Ardent-Apalone)?

Christywl, 2017-12-11:
Before that, I run the tests based on ros2/ros2@b12ebdc