RobotWebTools/rclnodejs #189: Change of API: hide ROSMessageWrapper type

kenny-y, 2017-11-22:
Purpose:
 - Remove `rclnodejs.require()` from customer-visible API
 - Use string directly to create publisher/subscriber/client/service
 - Provide a function that can create a blank message/request JavaScript object for later use

Example code:

```
  let obj = rclnodejs.createMessageObject('sensor_msgs/msg/JointState');
  obj.name = ['Alice', 'Bob'];
  obj.position = [0, 1];
  // obj.xxx = ...

  const publisher = node.createPublisher({package: 'sensor_msgs',
                                          type: 'msg',
                                          message: 'JointState'
                                          }, 'topic_name');
   -- OR --
  const publisher = node.createPublisher('sensor_msgs/msg/JointState', 'topic_name');

  publisher.publish({name: ['Alice', 'Bob'], position: [0, 1], /*...*/});
   -- OR --
  publisher.publish(obj);
```

minggangw, 2017-11-28:
That simplifies the invoking and makes it clear :+1:

qiuzhong, 2017-11-28:
@kenny-y , @minggangw , so what about the `client/service` case? Does this change also apply to the `node.createClient` and `node.createService` as well?

qiuzhong, 2017-11-28:
Sorry. I didn't see `Use string directly to create publisher/subscriber/client/service`.

minggangw, 2017-11-29:
Reopen this as 359541685 is only part of this feature.

kenny-y, 2017-11-29:
@qiuzhong Examples in description were updated to better illustrate the change...

qiuzhong, 2017-11-30:
Thanks, @kenny-y , so the usage of response in the callback of `client.sendRequest()` should be like this?
```
let result = response.template;
// other code
response.send(result);
```

kenny-y, 2017-11-30:
Yes, this response class is more clear in concept than returning an object, plus that providing an empty response object as parameter is informal.

The response object concept is common among server programming environments, e.g. ASP, JSP, ...

[image]

closed 2017-11-30 by kenny-y

qiuzhong, 2017-12-05:
Verified. Now these APIs are much easier and more intuitive to use.

PR #208 (2017-11-29): Adapt to an object as parameter when invoking require method
Currently we support two kinds of parameter to require a specific
message, like:
1.
// Require a package, including msg and srv.
let String = rclnodejs.require('std_msgs').msg.String;
2.
// Require a single message or service.
let String = rclnodejs.require('std_msgs/msg/String');

This patch enables to require a message by an object, e.g.

let String = rclnodejs.require({package: 'std_msgs', type: 'msg',
message: 'String'});

Fix #189
commit 41da929d (2017-11-29): Adapt to an object as parameter when invoking require method

Currently we support two kinds of parameter to require a specific
message, like:

1.
// Require a package, including msg and srv.
let String = rclnodejs.require('std_msgs').msg.String;
2.
// Require a single message or service.
let String = rclnodejs.require('std_msgs/msg/String');

This patch enables to require a message by an object, e.g.

let String = rclnodejs.require({package: 'std_msgs', type: 'msg',
message: 'String'});

Fix #189
PR #210 (2017-11-30): Change of API (Issue #189)
Fixes #189

Hide ROS message wrapper class in API, use simple string or plain JS object as parameter.

Also adding Response class to request callback of service.
commit 239acea9 (2017-11-30): Change of API (Issue #189) (#210)

* Fixes #189

Hide ROS message wrapper class in API, use simple string or plain JS object as parameter.
Also adding Response class to request callback of service.