RobotWebTools/rclnodejs #212: node.createTimer() or setInterval() from node.js

qiuzhong, 2017-11-30:
I feel like the `node.createTimer(period, callback)` behaves like the `setInterval()` from node.js. What's the difference between them? Should I use which one in the unit tests?

qiuzhong, 2017-11-30:
@minggangw , @kenny-y

minggangw, 2017-11-30:
The timer provided by `rclnodejs` is leveraging the one from [rcl](https://github.com/ros2/rcl/blob/master/rcl/include/rcl/timer.h#L63) lib, and it supports to set the interval's granularity by `nanoseconds` (`rclnodejs` supports millisecond now, because we have [limitation](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Number/MAX_SAFE_INTEGER) of maximum integer in JavaScript language, we may use string to represent an integer to change this).

Most of the platforms have a `timer` function, like `setInterval()` in JavaScript. I think ROS attempts to eliminate the fragmentation among different platforms (Linux/Windows/macOS). From engineering point of view, we just expose the ability from middle ware to the top layer. That is what I can observed.

kenny-y, 2017-12-01:
Adding a relevant point here: maybe we should enable 2 forms of `node.createTimer()` API, e.g.

```
1. node.createTimer(time, callbackFunction); // For ROS programmers
1. node.createTimer(callbackFunction, time); // For JavaScript programmers who might prefer this order
```

```
if (arguments.length === 2) {
  const t1 = typeof arguments[0];
  const t2 = typeof arguments[1];
  if (t1 === 'string' && t2 === 'function') {
    // (time, callbackFunction)
  } else if (t1 === 'function' && t2 === 'number') {
    // (callbackFunction, time)
  }
}
```

qiuzhong, 2017-12-05:
Thanks all. That's clear. I think we can close this issue.

closed 2017-12-05 by qiuzhong