RobotWebTools/rclnodejs #222: Cross-language interaction failed

minggangw, 2017-12-05:
Recently when running script:
```
node ./scripts/compile_cpp.js && node ./scripts/run_test.js
```

The appveyor ci fails consistently, please see pr #221, and I noticed that the failed cases are the ones which test the compatibility between `rclnodejs` and `rclpy`/`rclcpp` clients. We verify the messages by reading what has been written to the std out, which may cause some mistake when the console outputs more. e.g. If we enable the debug build, there will be a lot of debug messages which will also influence the test result.

@qiuzhong would you please change the strategy to verify the interaction related functions. Maybe we could create another `publisher/subscription` in `rclpy/rclcpp` to forward what has been `received/published` back to `rclnodejs` instead of reading from console output.

minggangw, 2017-12-05:
Using our development machine on Linux, all unit tests can pass. But the performance of the machine used by the ci may be not as high as ours, and we have to eliminate the cases which depend on the timing sequence.

qiuzhong, 2017-12-06:
@minggangw , indeed, by capturing the recipient standard output is not a perfect way. I wonder in 
`rclnodejs publisher -> rclpy/rclcpp subscription` case, how does the subscription send back the message to the rclnodejs publisher nodes? I don't quite understand how to add more `publisher/subscription` nodes in `rclpy/rclcpp` to forward messages?

minggangw, 2017-12-06:
We can adopt the policy below:

- In `rclnodejs`, create a publisher; In `rclpy`, create a subscription.
- The publisher of `rclnodejs` sends a message to subscription.
- The subscription receives the message from the `rclnodejs` side, then uses another publisher to send back what it just received.
- Again, in `rclnodejs`, create a subscription to receive message from `rclpy` to verify if the message is equal to what it just sent. 

This can improve the ut pass rate on ci, and not a functional requirement.

qiuzhong, 2017-12-07:
I tried the new strategy to verify the interaction between publisher/subscription and it works. However, there is another issue: the `rclpy` node is quite slow. If run the `rclnodejs` test cases as soon as I launch the `rclpy` node, the `rclpy` node often fails to response as it may not be initialized. So maybe I should still use the timer to send message from `rclnodejs` publisher to `rclpy` subscription and then send it back from `rclpy` publisher to `rclnodejs` subscription and compare the sent and received message.

qiuzhong, 2017-12-07:
The following unit test files are using the console output and need to use the new strategy for verification:
* test-cross-lang.js
* test-msg-type-cpp-node.js
* test-msg-type-py-node.js
* test-multi-nodes.js

minggangw, 2017-12-07:
I think we can take advantage of the `stdout` here. When the `rclpy` is ready, then send a specific text, and `rclnodejs` will listen to the output of `stdout` to check the existence of the keyword. Compared with the previous usage of `stdout`, we do not have to find the wording exactly.

qiuzhong, 2017-12-07:
You mean checking the output (like RegExp) rather than exact word comparison?

minggangw, 2017-12-07:
Yes, we don't need to do something like `deepEqual`.

qiuzhong, 2017-12-08:
I update the unit tests to use RegExp for checking the console output. This is for improving the CI pass rate temporarily.

minggangw, 2017-12-20:
Hi @qiuzhong how about the progress of this issue?

qiuzhong, 2017-12-20:
I'm afraid it's not as easy as I thought. The publisher/subscription model is just fine. But I got some block for the client/service model even on my Linux system. I'm fixing them.

qiuzhong, 2017-12-28:
This issue has not been completely fixed.

minggangw, 2017-12-28:
I notice that there is still frequent failure on mac/linux platforms, so I suggest reverting 5bf3230bcf first and reland when it gets fixed.

qiuzhong, 2017-12-29:
I'm confused. I cannot reproduced in my branch on the CI right now.

qiuzhong, 2017-12-29:
#256 all passed to fix this. But let's keep this issue for a while and see if the failure happen again in future.

closed 2018-01-02 by minggangw

minggangw, 2018-01-03:
During the recent commits, all the ci tests passed! It seems cba40f313 has resolved this stubborn problem, excellent job @qiuzhong !

PR #250 (2017-12-27): Use new strategy to verify message transmitting
Before we used console output of C++ and Python subscription to
check the message, this may sometimes fail on low performance machine.
Now we use a new strategy to verify message:
* Rclnodejs publisher sends message to C++ or Python subscriptions
* The subscriptions send the message back by creating a new publisher
* The rclnodejs side then creates a new subscription to receive messages

Therefore, there is no need to keep the rclnodejs subscription - C++/Python
publisher cases any more.

Fix: #222
commit 5bf3230b (2017-12-27): Use new strategy to verify message transmitting (#250)

* Use new strategy to verify message transmitting

Before we used console output of C++ and Python subscription to
check the message, this may sometimes fail on low performance machine.
Now we use a new strategy to verify message:
* Rclnodejs publisher sends message to C++ or Python subscriptions
* The subscriptions then send the message back by creating a new publisher
* The rclnodejs side then creates a new subscription to receive messages

Therefore, there is no need to keep the rclnodejs subscription - C++/Python
publisher cases any m [... cut]
commit cba40f31 (2018-01-02): Fix issue 222 (#256)

* Use new strategy to verify message transmitting

Before we used console output of C++ and Python subscription to
check the message, this may sometimes fail on low performance machine.
Now we use a new strategy to verify message:
* Rclnodejs publisher sends message to C++ or Python subscriptions
* The subscriptions then send the message back by creating a new publisher
* The rclnodejs side then creates a new subscription to receive messages

Therefore, there is no need to keep the rclnodejs subscription - C++/Python
publisher cases any more.

Fix: #222