RobotWebTools/rclnodejs #225: Publisher/Subscription blocked when a lot of messages are sending on Windows

qiuzhong, 2017-12-06:
We encountered the blocking between publisher and subscription when we tried to send 100k messages from publisher to subscription. It's not always happened but we encountered it once.

## How we encountered it?
* Launch a subscription
`node 01-one-subscription-100k.js`

* Launch a publisher to send 100k messages of `std_msgs/msg/String`
`node 01-one-publisher-100k.js`

The test case is from [here](https://github.com/qiuzhong/web-test-suite/blob/rclnodejs-stability/ros/rclnodejs/stability/)

qiuzhong, 2017-12-06:
Both of them stopped at 38654 sending.

minggangw, 2017-12-06:
For the memory usage, the cost is on a normal level. When encountering this specific case next time, can we launch a new publisher/subscription, which ensures to work properly,  to identify which end (subscription/publisher) is hanging.

e.g. for one case, the publisher is dead, but the subscription call still receive messages from another publisher by the same topic.

closed 2019-06-26 by minggangw