RobotWebTools/rclnodejs #231: Verify on all platforms: ARM/x86, 32/64bit

kenny-y, 2017-12-12:
We're doing development on 64bit ia64 on Win/Linux/Mac... we don't have proof that rclnodejs works on other supported platforms yet.

qiuzhong, 2017-12-12:
@kenny-y , I thought you mean `x86-64` or `amd64`, which is totally different with `ia64`.  Now `IA32`  32/64bit are supported and not tested on ARM platform.

kenny-y, 2017-12-12:
Yes ia64 is a typo 👍

minggangw, 2017-12-12:
I have checked the document of the three current ci we use, non of them supports the arm arch.

kenny-y, 2017-12-13:
Let's keep this open and look for solution in the long run :)

ros2jsguy, 2020-09-03:
Hmm. I just created a small test project that installs, builds and runs with no issues.

- raspberry 4, 8Mb
- ubuntu 20.04
- ros2 foxy distro
- node 12.18.3
- yarn 1.22.5
- rclnodejs 0.15.3

I've been running this configuration for about a week but primarily using npm. I just installed yarn trying to replicate your report. Can you share any additional details?

MaximeAubanel, 2020-09-03:
I feel that the log start to make sense for the `rclnodejs` part but if you wantthe full log I can send it also.

```
gyp info using node-gyp@5.1.0
gyp info using node@12.18.3 | linux | arm64
gyp info find Python using Python version 2.7.18 found at "/usr/bin/python2"
gyp http GET https://nodejs.org/download/release/v12.18.3/node-v12.18.3-headers.tar.gz
gyp http 200 https://nodejs.org/download/release/v12.18.3/node-v12.18.3-headers.tar.gz
gyp http GET https://nodejs.org/download/release/v12.18.3/SHASUMS256.txt
gyp http 200 https://nodejs.org/download/release/v12.18.3/SHASUMS256.txt
gyp info spawn /usr/bin/python2
gyp info spawn args [
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/workspace/node_modules/int64-napi/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.cache/node-gyp/12.18.3/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/12.18.3',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/12.18.3/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/workspace/node_modules/int64-napi',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/workspace/node_modules/int64-napi/build'
  CC(target) Release/obj.target/int64/src/int64.o
  SOLINK_MODULE(target) Release/obj.target/int64.node
  COPY Release/int64.node
[3/3] ⠠ rclnodejs
[-/3] ⠠ waiting...
verbose 74.418493883 gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@12.18.3 | linux | arm64
gyp info find Python using Python version 2.7.18 found at "/usr/bin/python2"
gyp info spawn /usr/bin/python2
gyp info spawn args [
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/workspace/node_modules/rclnodejs/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.cache/node-gyp/12.18.3/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/12.18.3',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/12.18.3/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/workspace/node_modules/rclnodejs',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
[... 81 lines cut ...]
Command: node scripts/generate_messages.js
Arguments: 
Directory: /home/workspace/node_modules/rclnodejs
Output:
Start JavaScript message generation...
Generation complete.
Segmentation fault (core dumped)
    at ProcessTermError.ExtendableBuiltin (/usr/lib/node_modules/yarn/lib/cli.js:721:66)
    at ProcessTermError.MessageError (/usr/lib/node_modules/yarn/lib/cli.js:750:123)
    at new ProcessTermError (/usr/lib/node_modules/yarn/lib/cli.js:790:113)
    at ChildProcess.<anonymous> (/usr/lib/node_modules/yarn/lib/cli.js:25884:17)
    at ChildProcess.emit (events.js:315:20)
    at maybeClose (internal/child_process.js:1021:16)
    at Socket.<anonymous> (internal/child_process.js:443:11)
    at Socket.emit (events.js:315:20)
    at Pipe.<anonymous> (net.js:674:12)
error /home/workspace/node_modules/rclnodejs: Command failed.
Exit code: 139
Command: node scripts/generate_messages.js
Arguments: 
Directory: /home/workspace/node_modules/rclnodejs
Output:
Start JavaScript message generation...
Generation complete.
```

ros2jsguy, 2020-09-03:
That's helpful. 
```
Start JavaScript message generation...
Generation complete.
Segmentation fault (core dumped)
    at ProcessTermError.ExtendableBuiltin (/usr/lib/node_modules/yarn/lib/cli.js:721:66)
    at ProcessTermError.MessageError (/usr/lib/node_modules/yarn/lib/cli.js:750:123)
    at new ProcessTermError (/usr/lib/node_modules/yarn/lib/cli.js:790:113)
    at ChildProcess.<anonymous> (/usr/lib/node_modules/yarn/lib/cli.js:25884:17)
    at ChildProcess.emit (events.js:315:20)
    at maybeClose (internal/child_process.js:1021:16)
    at Socket.<anonymous> (internal/child_process.js:443:11)
    at Socket.emit (events.js:315:20)
    at Pipe.<anonymous> (net.js:674:12)
error /home/workspace/node_modules/rclnodejs: Command failed.
Exit code: 139
Command: node scripts/generate_messages.js
```
I have seen this segfault issue on my mac workstation a few times but hard to replicate consistently.  Please try uninstalling and then reinstalling rclnodejs. That has worked for me in such situations. Sorry you hit this glitch. We can look into this issue more closely.

MaximeAubanel, 2020-09-03:
The thing is it's in a docker container. For the sake of dev purpose,  every time it launch it yarn install. So it is re installing rclnodejs. And it keeps coming up. Actually, it did compile correctly one time when I build the first time with --verbose

ros2jsguy, 2020-09-03:
@MaximeAubanel  :( bummer this is blocking you. I'm trying to replicate and isolate the issue in my rclnodejs dev env now.

MaximeAubanel, 2020-09-03:
Have you found anything interesting ?

ros2jsguy, 2020-09-03:
Making progress. Had to setup lldb and llnode to get deep insight into the core dump. I have not isolated the issue fully yet but circling in. Will update as soon as I know something definitive.

MaximeAubanel, 2020-09-03:
I don't know if there is a link.

But when I'm trying to yarn install directly inside the docker through a `RUN` call it doesn't compile, `amd64` or `arm64`.

There is an error coming from `gyp` on command `node-gyp rebuild` used by `rclnodejs`.

```
error /home/workspace/node_modules/rclnodejs: Command failed.
Exit code: 1
Command: node-gyp rebuild
Arguments: 
Directory: /home/workspace/node_modules/rclnodejs
Output:
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@12.18.3 | linux | x64
gyp info find Python using Python version 2.7.18 found at "/usr/bin/python"
gyp info spawn /usr/bin/python
gyp info spawn args [
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/workspace/node_modules/rclnodejs/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.cache/node-gyp/12.18.3/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/12.18.3',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/12.18.3/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/workspace/node_modules/rclnodejs',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
[eval]:1
console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')
                                          ^

TypeError: Cannot read property 'replace' of undefined
    at [eval]:1:43
    at Script.runInThisContext (vm.js:120:18)
    at Object.runInThisContext (vm.js:309:38)
    at Object.<anonymous> ([eval]-wrapper:10:26)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at evalScript (internal/process/execution.js:94:25)
    at internal/main/eval_string.js:23:3
gyp: Call to 'node -e "console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')"' returned exit status 1 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.19.76-linuxkit
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/workspace/node_modules/rclnodejs
gyp ERR! node -v v12.18.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
```

ros2jsguy, 2020-09-03:
I have a work-around (hack) that is working consistently in my testing. I'll submit a PR shortly.

MaximeAubanel, 2020-09-04:
Thank you @ros2jsguy 🚀

closed 2025-06-30 by minggangw