RobotWebTools/rclnodejs #232: Failed to Assign a New Value to an Array Element

kenny-y, 2017-12-12:
The following test case would fail

``` JavaScript
    const r = Math.round(Math.random() * 100);
    msg1.data[0] = r;
    console.log(r, msg1.data[0], msg2.data[0], o);
    assert(msg1.data[0] == r);
```

The root cause is that the getter of data property is wrong. We should cache an array in `this` object, clear that cache array when it's necessary, and generate the cache array when requested in `.data` getter when it's empty.

e.g. This the generated code for Int64MultiArray
``` JavaScript
  get data() {
    let values = [];
    this._wrapperFields['data'].data.forEach((wrapper, index) => {
      values.push(wrapper.data);
    });
    return values;
  }
```

Resolution
``` JavaScript
  get data() {
    if (!this.dataArray) {
      this.dataArray = [];
      this._wrapperFields['data'].data.forEach((wrapper, index) => {
        this.dataArray.push(wrapper.data);
      });
    }
    return this.dataArray;
  }

// TODO: assign undefined to this.dataArray when necessary.
```

TypedArray implementation doesn't have this issue so far.

kenny-y, 2017-12-12:
@minggangw Would you please look into this? I suppose we should clear `this.dataArray` when doing `.copyRefObject()`, and we should copy `this.dataArray` to `this._wrapperFields['data']` when necessary... is there anything else?

minggangw, 2017-12-12:
I will take this.

PR #248 (2017-12-25): Use a number variable of message class to hold the array element
When a message which has an array property, user can get it by the get
method, meanwhile any element in the array can be re-assigned by
operator = through index. But the array gotten by the get method is generated as a
temporary variable and will not influence the one which is stored in
the original message.

This patch creates a member variable to hold the property, which has a
type of primitive array.

Fix #232
commit 3bfc0b28 (2017-12-25): Use mumber variable of message class to hold the array element

When a message which has an array property, user can get it by the get
method, meanwhile any element in the array can be re-assigned by
operator = through index. But the array gotten by the get method is generated as a
temporary variable and will not influence the one which is stored in
the original message.

This patch creates a member variable to hold the property, which has a
type of primitive array.

Fix #232
closed 2017-12-25 by minggangw