RobotWebTools/rclnodejs #243: Undefined properties of requests are converted to 0 in the callback of createService

qiuzhong, 2017-12-18:
Let's say the client sends an ill-formatted request `{a:1}` for the service `example_interfaces/srv/AddTwoInts` (which requires two properties in the request), in the callback of 
`createService`:
```
  node.createService('example_interfaces/srv/AddTwoInts', 'add_two_ints', (request, response) => {
    console.log(`Incoming request: ${typeof request}`, request);
    let result = response.template;
    result.sum = request.a + request.b;
    console.log(`Sending response: ${typeof result}`, result, '\n--');
    response.send(result);
  });
```
We get this output:
```
Incoming request: object { a: 69, b: 0 }
Sending response: object { sum: 69 } 
--

```

Undefined property `request.b` is converted to 0 and doesn't throw an Error. If you print the type of request.b, it's `number`.

qiuzhong, 2017-12-18:
@kenny-y , @minggangw , what do you think?

minggangw, 2017-12-20:
Sorry for the late response, when we create a JavaScript object in `rclnodejs`, and a c struct memory will be allocated simultaneously, which is done by [ref](https://www.npmjs.com/package/ref). So, I think why you get `b == 0` is that the `ref` sets default value for the members of the struct, even though there is no corresponding property in JavaScript object.

qiuzhong, 2017-12-20:
So is this the well-defined behavior for `rclnodejs`? If so, I don't think it's a robust service to accept ill-formatted requests.

kenny-y, 2017-12-20:
I think this and #241 can fall into the same category -- strict structure/type checking before using the object

qiuzhong, 2018-02-26:
Fixed by #274

PR #274 (2018-02-26): Check the consistency of the message's members when being serialized
Currently when a message is instantiated, the values of its data members will
be a "default". Reference the issue 243:

If we have a request which will be sent to the service to calculate the
sum of two integers, and create {a: 1} instead of {a: 1, b: 1} which
means we do not assign the value for member b explicitly. What the
service will receive is {a: 1, b: 0} from the client request. We may
just forget the assignment unintentionally, but rclnodejs should produce
an error.

This patch implements the strategy below:
1. If th [... cut]
commit 925966b6 (2018-02-26): Check the consistency of the message's members when being serialized

Currently when a message is instantiated, the values of its data members will
be a "default". Reference the issue 243:

If we have a request which will be sent to the service to calculate the
sum of two integers, and create {a: 1} instead of {a: 1, b: 1} which
means we do not assign the value for member b explicitly. What the
service will receive is {a: 1, b: 0} from the client request. We may
just forget the assignment unintentionally, but rclnodejs should produce
an error.

This patch implements the strategy below:
1. If t [... cut]
closed 2018-02-26 by minggangw