RobotWebTools/rclnodejs #260: Error in npm install

prasun2712, 2018-01-23:
> rclnodejs@0.3.1 install /home/username/rclnodejs
> node-gyp rebuild

make: Entering directory '/home/username/rclnodejs/build'
  CXX(target) Release/obj.target/rclnodejs/src/addon.o
cc1plus: error: /home/username/ros2_ws/install_isolated/common_interfaces:/home/username/ros2_ws/install_isolated/visualization_msgs:/home/username/ros2_ws/install_isolated/rviz_default_plugins:/home/username/ros2_ws/install_isolated/rviz2:/home/username/ros2_ws/install_isolated/rviz_common:/home/username/ros2_ws/install_isolated/dummy_robot_bringup:/home/username/ros2_ws/install_isolated/robot_state_publisher:/home/username/ros2_ws/install_isolated/kdl_parser:/home/username/ros2_ws/install_isolated/urdf:/home/username/ros2_ws/install_isolated/urdfdom_headers:/home/username/ros2_ws/install_isolated/urdfdom:/home/username/ros2_ws/install_isolated/uncrustify:/home/username/ros2_ws/install_isolated/trajectory_msgs:/home/username/ros2_ws/install_isolated/topic_monitor:/home/username/ros2_ws/install_isolated/pendulum_control:/home/username/ros2_ws/install_isolated/tlsf_cpp:/home/username/ros2_ws/install_isolated/tlsf:/home/username/ros2_ws/install_isolated/tinyxml_vendor:/home/username/ros2_ws/install_isolated/pluginlib:/home/username/ros2_ws/install_isolated/tinyxml2_vendor:/home/username/ros2_ws/install_isolated/tf2_geometry_msgs:/home/username/ros2_ws/install_isolated/tf2_eigen:/home/username/ros2_ws/install_isolated/tf2_ros:/home/username/ros2_ws/install_isolated/tf2_msgs:/home/username/ros2_ws/install_isolated/tf2:/home/username/ros2_ws/install_isolated/test_security:/home/username/ros2_ws/install_isolated/test_rclcpp:/home/username/ros2_ws/install_isolated/test_communication:/home/username/ros2_ws/install_isolated/test_msgs:/home/username/ros2_ws/install_isolated/stereo_msgs:/home/username/ros2_ws/install_isolated/std_srvs:/home/username/ros2_ws/install_isolated/shape_msgs:/home/username/ros2_ws/install_isolated/ros1_bridge:/home/username/ros2_ws/install_isolated/intra_process_demo:/home/username/ros2_ws/install_isolated/image_tools:/home/username/ros2_ws/install_isolated/dummy_sensors:/home/username/ros2_ws/install_isolated/demo_nodes_cpp:/home/username/ros2_ws/install_isolated/sensor_msgs:/home/username/ros2_ws/install_isolated/lifecycle:/home/username/ros2_ws/install_isolated/rclcpp_lifecycle:/home/username/ros2_ws/install_isolated/rclpy:/home/username/ros2_ws/install_isolated/logging_demo:/home/username/ros2_ws/install_isolated/examples_rclcpp_minimal_timer:/home/username/ros2_ws/install_isolated/examples_rclcpp_minimal_subscriber:/home/username/ros2_ws/install_isolated/examples_rclcpp_minimal_service:/home/username/ros2_ws/install_isolated/examples_rclcpp_minimal_publisher:/home/username/ros2_ws/install_isolated/examples_rclcpp_minimal_composition:/home/username/ros2_ws/install_isolated/examples_rclcpp_minimal_client:/home/username/ros2_ws/install_isolated/dummy_map_server:/home/username/ros2_ws/install_isolated/demo_nodes_cpp_native:/home/username/ros2_ws/install_isolated/composition:/home/username/ros2_ws/install_isolated/rclcpp:/home/username/ros2_ws/install_isolated/rcl_lifecycle:/home/username/ros2_ws/install_isolated/rcl:/home/username/ros2_ws/install_isolated/nav_msgs:/home/username/ros2_ws/install_isolated/diagnostic_msgs:/home/username/ros2_ws/install_isolated/geometry_msgs:/home/username/ros2_ws/install_isolated/actionlib_msgs:/home/username/ros2_ws/install_isolated/std_msgs:/home/username/ros2_ws/install_isolated/sros2:/home/username/ros2_ws/install_isolated/rviz_yaml_cpp_vendor:/home/username/ros2_ws/install_isolated/rviz_rendering_tests:/home/username/ros2_ws/install_isolated/rviz_rendering:/home/username/ros2_ws/install_isolated/rviz_ogre_vendor:/home/username/ros2_ws/install_isolated/rviz_assimp_vendor:/home/username/ros2_ws/install_isolated/rttest:/home/username/ros2_ws/install_isolated/rcl_interfaces:/home/username/ros2_ws/install_isolated/pendulum_msgs:/home/username/ros2_ws/install_isolated/lifecycle_msgs:/home/username/ros2_ws/install_isolated/example_interfaces:/home/username/ros2_ws/install_isolated/builtin_interfaces:/home/username/ros2_ws/install_isolated/rosidl_default_runtime:/home/username/ros2_ws/install_isolated/rosidl_default_generators:/home/username/ros2_ws/install_isolated/rosidl_typesupport_cpp:/home/username/ros2_ws/install_isolated/rosidl_generator_py:/home/username/ros2_ws/install_isolated/rosidl_typesupport_c:/home/username/ros2_ws/install_isolated/rmw_opensplice_cpp:/home/username/ros2_ws/install_isolated/rosidl_typesupport_opensplice_c:/home/username/ros2_ws/install_isolated/rosidl_typesupport_opensplice_cpp:/home/username/ros2_ws/install_isolated/rmw_implementation:/home/username/ros2_ws/install_isolated/rmw_fastrtps_cpp:/home/username/ros2_ws/install_isolated/rosidl_typesupport_introspection_cpp:/home/username/ros2_ws/install_isolated/rosidl_typesupport_introspection_c:/home/username/ros2_ws/install_isolated/rmw_connext_cpp:/home/username/ros2_ws/install_isolated/rosidl_typesupport_connext_c:/home/username/ros2_ws/install_isolated/rosidl_typesupport_connext_cpp:/home/username/ros2_ws/install_isolated/rosidl_generator_cpp:/home/username/ros2_ws/install_isolated/rmw:/home/username/ros2_ws/install_isolated/rosidl_generator_c:/home/username/ros2_ws/install_isolated/rosidl_typesupport_interface:/home/username/ros2_ws/install_isolated/rosidl_generator_dds_idl:/home/username/ros2_ws/install_isolated/rosidl_cmake:/home/username/ros2_ws/install_isolated/rosidl_parser:/home/username/ros2_ws/install_isolated/ros2topic:/home/username/ros2_ws/install_isolated/ros2srv:/home/username/ros2_ws/install_isolated/ros2service:/home/username/ros2_ws/install_isolated/ros2run:/home/username/ros2_ws/install_isolated/ros2pkg:/home/username/ros2_ws/install_isolated/ros2node:/home/username/ros2_ws/install_isolated/ros2msg:/home/username/ros2_ws/install_isolated/ros2cli:/home/username/ros2_ws/install_isolated/rmw_implementation_cmake:/home/username/ros2_ws/install_isolated/rmw_connext_shared_cpp:/home/username/ros2_ws/install_isolated/resource_retriever:/home/username/ros2_ws/install_isolated/rcutils:/home/username/ros2_ws/install_isolated/python_cmake_module:/home/username/ros2_ws/install_isolated/class_loader:/home/username/ros2_ws/install_isolated/poco_vendor:/home/username/ros2_ws/install_isolated/osrf_pycommon:/home/username/ros2_ws/install_isolated/orocos_kdl:/home/username/ros2_ws/install_isolated/opensplice_cmake_module:/home/username/ros2_ws/install_isolated/libcurl_vendor:/home/username/ros2_ws/install_isolated/launch_testing:/home/username/ros2_ws/install_isolated/launch:/home/username/ros2_ws/install_isolated/ament_index_cpp:/home/username/ros2_ws/install_isolated/ament_cmake_gmock:/home/username/ros2_ws/install_isolated/ament_cmake_gtest:/home/username/ros2_ws/install_isolated/gtest_vendor:/home/username/ros2_ws/install_isolated/gmock_vendor:/home/username/ros2_ws/install_isolated/fastrtps_cmake_module:/home/username/ros2_ws/install_isolated/fastrtps:/home/username/ros2_ws/install_isolated/fastcdr:/home/username/ros2_ws/install_isolated/examples_rclpy_minimal_subscriber:/home/username/ros2_ws/install_isolated/examples_rclpy_minimal_service:/home/username/ros2_ws/install_isolated/examples_rclpy_minimal_publisher:/home/username/ros2_ws/install_isolated/examples_rclpy_minimal_client:/home/username/ros2_ws/install_isolated/examples_rclpy_executors:/home/username/ros2_ws/install_isolated/demo_nodes_py:/home/username/ros2_ws/install_isolated/console_bridge:/home/username/ros2_ws/install_isolated/connext_cmake_module:/home/username/ros2_ws/install_isolated/ament_cmake_ros:/home/username/ros2_ws/install_isolated/ament_lint_common:/home/username/ros2_ws/install_isolated/ament_cmake_uncrustify:/home/username/ros2_ws/install_isolated/ament_uncrustify:/home/username/ros2_ws/install_isolated/ament_tools:/home/username/ros2_ws/install_isolated/ament_cmake_pyflakes:/home/username/ros2_ws/install_isolated/ament_pyflakes:/home/username/ros2_ws/install_isolated/ament_cmake_pep8:/home/username/ros2_ws/install_isolated/ament_pep8:/home/username/ros2_ws/install_isolated/ament_cmake_pep257:/home/username/ros2_ws/install_isolated/ament_pep257:/home/username/ros2_ws/install_isolated/ament_lint_auto:/home/username/ros2_ws/install_isolated/ament_cmake_pytest:/home/username/ros2_ws/install_isolated/ament_cmake_nose:/home/username/ros2_ws/install_isolated/ament_cmake_flake8:/home/username/ros2_ws/install_isolated/ament_cmake_cpplint:/home/username/ros2_w [... cut ...]
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs rclnodejs
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls rclnodejs
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/username/rclnodejs/npm-debug.log

minggangw, 2018-01-24:
I suggest using the binary package of ROS2, you can download from http://ci.ros2.org/view/packaging/job/packaging_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2, or do not build by `--isolated`

minggangw, 2018-01-24:
Also I prefer upgrading your nodejs to LTS carbon (v8.9.4), we will point out the required nodejs version ASAP in README, thanks!

minggangw, 2018-01-29:
As f628b9a5 and 1d8c49e are landed, the problem should be resolved and close this thread.

closed 2018-01-29 by minggangw