RobotWebTools/rclnodejs #267: Error in npm install on the instance Google Cloud Platform

wchuchnowski, 2018-02-09:
wchuchnowski@instance-1:~/ros2_install/rclnodejs$ node-gyp rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@3.6.2
gyp info using node@6.11.4 | linux | x64
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp info spawn /usr/bin/python2
gyp info spawn args [ '/home/wchuchnowski/.nvm/versions/node/v6.11.4/lib/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/wchuchnowski/ros2_install/rclnodejs/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/wchuchnowski/.nvm/versions/node/v6.11.4/lib/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/wchuchnowski/.node-gyp/6.11.4/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/wchuchnowski/.node-gyp/6.11.4',
gyp info spawn args   '-Dnode_gyp_dir=/home/wchuchnowski/.nvm/versions/node/v6.11.4/lib/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/wchuchnowski/.node-gyp/6.11.4/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/wchuchnowski/ros2_install/rclnodejs',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/home/wchuchnowski/ros2_install/rclnodejs/build'
CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
  CXX(target) Release/obj.target/rclnodejs/src/handle_manager.o
  CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
In file included from ../src/rcl_bindings.cpp:33:0:
../src/rcl_bindings.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE rclnodejs::TimerGetTimeSinceLastCall(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rcl_bindings.cpp:150:74: error: invalid conversion from ‘int64_t* {aka long int*}’ to ‘uint64_t* {aka long unsigned int*}’ [-fpermissive]
       RCL_RET_OK, rcl_timer_get_time_since_last_call(timer, &elapsed_time),
                                                                          ^
../src/macros.hpp:20:16: note: in definition of macro ‘CHECK_OP_AND_THROW_ERROR_IF_NOT_TRUE’
     if (lhs op rhs) { \
                ^
../src/rcl_bindings.cpp:149:3: note: in expansion of macro ‘THROW_ERROR_IF_NOT_EQUAL’
   THROW_ERROR_IF_NOT_EQUAL(
   ^
In file included from /home/wchuchnowski/ros2_install/ros2-linux/include/rcl/wait.h:31:0,
                 from /home/wchuchnowski/ros2_install/ros2-linux/include/rcl/rcl.h:81,
                 from ../src/rcl_bindings.cpp:20:
/home/wchuchnowski/ros2_install/ros2-linux/include/rcl/timer.h:314:1: note:   initializing argument 2 of ‘rcl_ret_t rcl_timer_get_time_since_last_call(const rcl_timer_t*, uint64_t*)’
 rcl_timer_get_time_since_last_call(const rcl_timer_t * timer, uint64_t * time_since_last_call);
 ^
rclnodejs.target.mk:117: recipe for target 'Release/obj.target/rclnodejs/src/rcl_bindings.o' failed
make: *** [Release/obj.target/rclnodejs/src/rcl_bindings.o] Error 1
make: Leaving directory '/home/wchuchnowski/ros2_install/rclnodejs/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/wchuchnowski/.nvm/versions/node/v6.11.4/lib/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Linux 4.13.0-1008-gcp
gyp ERR! command "/home/wchuchnowski/.nvm/versions/node/v6.11.4/bin/node" "/home/wchuchnowski/.nvm/versions/node/v6
.11.4/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/wchuchnowski/ros2_install/rclnodejs
gyp ERR! node -v v6.11.4
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok

qiuzhong, 2018-02-11:
It seems your `rcl` package is not recent. See [ros2/rcl@44801b3](https://github.com/ros2/rcl/commit/44801b398e6f053eb6b474d2caf9bb5319e3add3) and #266 keeps aligned with the this change. Please update `rcl` package and rebuild it.

minggangw, 2018-02-11:
Hi @wchuchnowski as our module `rclnodejs` is built on the latest [master](https://github.com/ros2/ros2/blob/master/ros2.repos) branch of ROS 2, the code may change. The error you met is due to the change of the interface of timer in `rcl` recently (You could reference @qiuzhong 's comment above).

I suggest you can download the binary package from the [ci.ros2.org](http://ci.ros2.org/view/packaging/job/packaging_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2) for Linux platform, also you can find the packages for Windwos and macOS (actually our CIs cover all these three platforms already).

I have a question here, I am curious what is your scenario of your application on the Google Cloud Platform. We want to collect the common cases in order to improve our performance and optimize the functions, thanks!

minggangw, 2018-02-11:
The [ROS 2 Ardent Apalone - 2018-02-09](https://github.com/ros2/ros2/releases/tag/release-ardent-20180209) was released a day ago, which also resolve the error above.

wchuchnowski, 2018-02-11:
Thanks for your help. I installed ros2 from master source and it works. Great thx

closed 2018-02-11 by wchuchnowski