RobotWebTools/rclnodejs #286: Installation not working with an ament overlay ws

vincentrou, 2018-03-09:
When calling npm install with `$AMENT_PREFIX_PATH = /home/vincentrou/ros2_ws/test_ws/install:/opt/ros/ardent/`
Here is the error : 
```
  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
../src/executor.cpp:17:32: fatal error: rcl/error_handling.h: No such file or directory
compilation terminated.
rclnodejs.target.mk:118: recipe for target 'Release/obj.target/rclnodejs/src/executor.o' failed
```
The file `build/rclnodejs.target.mk` contains a wrong include path : 
`-I/home/vincentrou/ros2_ws/test_ws/install:/opt/ros/ardent/include`
Instead of : 
`-I/home/vincentrou/ros2_ws/test_ws/install/include`
`-I/opt/ros/ardent/include`

minggangw, 2018-03-12:
The root cause here is you install two copies of `ROS2`, one is built from source and the other is from binary package (I guessed that from the content of  ```$AMENT_PREFIX_PATH``` variable you wrote), and when compiling the node.js module, the ```$AMENT_PREFIX_PATH``` will be considered as only one path (but here you had two) which triggered the compiling error.

vincentrou, 2018-03-12:
> The root cause here is you install two copies of ROS2

I do not have two copies of ROS2. ROS2 is only installed in `/opt/ros/ardent/`.
My overlay workspace `/home/vincentrou/ros2_ws/test_ws` only contains some custom messages that I want to use with rclnodejs.
So the $AMENT_PREFIX_PATH value (`/home/vincentrou/ros2_ws/test_ws/install:/opt/ros/ardent/`) seems perfectly normal.
I manage to make it work by calling `node-gyp build` instead of `node-gyp rebuild`. And by modifying the `build/rclnodejs.target.mk` file with the following includes and lib : 
`-I/home/vincentrou/ros2_ws/test_ws/install/include`
`-I/opt/ros/ardent/include`
`-L/home/vincentrou/ros2_ws/test_ws/install/lib`
`-L/opt/ros/ardent/lib`

But this is a dirty fix. A true fix would be to change the `binding.gyp` file to handle multiple ament workspace.

minggangw, 2018-03-12:
I see, I am going to fix this issue to make the compiling successful.

minggangw, 2018-03-13:
@qiuzhong would you please help verify this issue (you can create a simple ROS2 application by `ament` and source its `local_setup.bash`, then build the `rclnodejs`. I have verified Linux&Windows platform). If it is fixed, I will close this issue.

@vincentrou would you `npm install` from github to see if it has been fixed, thanks for your investigation.

vincentrou, 2018-03-13:
It works for me, thanks.

qiuzhong, 2018-03-14:
@minggangw , I tried on Windows and Linux. The installation issue was fixed. But in that case, I found another issue, see #291

minggangw, 2018-03-14:
So let's close this issue as it has been fixed and keep the #291 open.

PR #288 (2018-03-13): Expand the AMENT_PREFIX_PATH string in a list
When the AMENT_PREFIX_PATH variable contains multi-path, we have to
expand them in a list context to ensure the compiler can find the
headers and libraries.

Fix #286
commit 3bd57a7f (2018-03-13): Expand the AMENT_PREFIX_PATH string in a list

When the AMENT_PREFIX_PATH variable contains multi-path, we have to
expand them in a list context to ensure the compiler can find the
headers and libraries.

Fix #286
closed 2018-03-14 by minggangw