RobotWebTools/rclnodejs #291: [Reg] Unit tests got a lot of failures when AMENT_PREFIX_PATH is overrided

qiuzhong, 2018-03-14:
This issue is triggered from #286 . 

That is when the ROS2 environment variable `AMENT_PREFIX_PATH` is overrided with some other path plus the real `ros2/install` path, the install error of #286 is fixed, but it raises a lot of failures when running the unit tests. Even the unit test suite `test-message-translator` failed.

After replace the `AMENT_PREFIX_PATH` to a single `ros2/install` path, this issue disappears.

qiuzhong, 2018-03-14:
@minggangw , I think we simply should not override the `AMENT_PREFIX_PATH`.

minggangw, 2018-03-14:
Thanks, I will check it soon.

minggangw, 2018-03-14:
Hi @qiuzhong would you please verify this issue under the situation when the `AMENT_PREFIX_PATH` contains more than one path, thanks!

closed 2018-03-14 by minggangw

qiuzhong, 2018-03-14:
@minggangw , with PR293 most failures above are fixed. But there are still some test cases failed due to they will use the `AMENT_PREFIX_PATH` to find the ROS2 application like  `talker`. However, with multiple paths, the pathname of these applications are invalid.

qiuzhong, 2018-03-14:
In such case, I don't know which path is the "real" path of ROS2 environment. Basically, this issue will not happened on the CI. I don't think we need to modify the test cases for this specific condition.

minggangw, 2018-03-15:
I agree, as we will not meet multi-path case on CIs, let's mark this issue as fixed. Thanks!

PR #293 (2018-03-14): Wait for the generator to complete its work
When the variable AMENT_PREFIX_PATH is a list which contains multi-path,
the generator must wait all promises to be resolved, which means the
generator will search every possible package path and generate the
JavaScript files based on the .msg/.srv files.

This patch implemented the logic above.

Fix #291
commit f45c0544 (2018-03-14): Wait for the generator to complete its work

When the variable AMENT_PREFIX_PATH is a list which contains multi-path,
the generator must wait all promises to be resolved, which means the
generator will search every possible package path and generate the
JavaScript files based on the .msg/.srv files.

This patch implemented the logic above.

Fix #291