RobotWebTools/rclnodejs #313: v0.4.0 release

minggangw, 2018-04-12:
With the landing of [logging utility](https://github.com/RobotWebTools/rclnodejs/issues/281) and [benchmark tests](https://github.com/RobotWebTools/rclnodejs/issues/280), I think it is the proper opportunity to release v0.4.0 which will be the first stable release of rclnodejs (we have had 8 alpha&beta releases).

Meanwhile, I hope we can illustrate and compare the performance of rclnodejs with others (rclcpp&rclpy) through the statistics of PnP tests.

@Honry @hanyia @qiuzhong please append issues or patches you want to add into this release.

Honry, 2018-04-12:
@minggangw, great achievement! I think we should do a full testing before release.

qiuzhong, 2018-04-12:
And this full testing should also contain the protocol tests in `ros2-web-bridge` as a crash bug was found in `rclnodejs` when running ros2-web-bridge tests.

minggangw, 2018-04-13:
I will create a check list to track things we want to do in this release. Please edit that list if you have any requirement.

minggangw, 2018-07-26:
[v0.4.0](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.4.0) has been published, we will add the PnP result later.

closed 2018-07-26 by minggangw