RobotWebTools/rclnodejs #315: Add guard condition feature

minggangw, 2018-04-13:
We should implement the guard condition based on [rcl](https://github.com/ros2/rcl/blob/master/rcl/include/rcl/guard_condition.h#L32) library.

mattrichard, 2020-02-11:
@minggangw If you're not working on this I might take a stab at it. It looks relatively simple to implement, but the actual usage is more difficult to understand since there's very little documentation about guard conditions.

minggangw, 2020-02-12:
I'm not working on it, please go ahead. I'd like to share some my understandings of this feature:

guard condition acts as a sort of `EventEmitter` in Node.js, offering the event-driven mechanism in ROS2. One difficult point I remember is that we have to find out which guard has been triggered and pass it to the main thread. (In the current design, we don't get the entities that need to be handled)

minggangw, 2020-02-12:
Hi @mattrichard we have a slack channel at https://rclnodejsdev.slack.com/ created by @wayneparrott (thanks for creating it), you could join in and I think it will facilitate our daily development. I have sent the invitation to @koonpeng already 😄

mattrichard, 2020-02-12:
@minggangw if you could send an invitation to matthew.w.richard@gmail.com that would be awesome.

minggangw, 2020-02-12:
> if you could send an invitation to matthew.w.richard@gmail.com that would be awesome.

Done

PR #561 (2020-02-15): Add support for guard conditions
Implements guard conditions in rclnodejs. In addition, now only entities that are ready to be handled will be executed. This was initially needed since guard conditions should only run when triggered, but was extended to all entities. The approach taken was passing all RclHandles that are ready to be executed back to the main process.

All changes to the .cpp files should be carefully reviewed for quality, thread safety, and alignment with future needs. There's a lot of looping occurring now, but this is not much different than the  rclpy implementation.

Fix # [... cut]
commit e76626ca (2020-02-15): Add support for guard conditions (#561)

Implements guard conditions in rclnodejs. In addition, now only entities that are ready to be handled will be executed. This was initially needed since guard conditions should only run when triggered, but was extended to all entities. The approach taken was passing all RclHandles that are ready to be executed back to the main process.

Fix #315
closed 2020-02-15 by minggangw