RobotWebTools/rclnodejs #351: [rosidl_gen]Accept a plain object to be assigned to the property of a message

minggangw, 2018-06-04:
For the current JavaScript messages generated by `rosidl_gen`, if the property is another kind of message, e.g. `std_msgs/Header`

```
builtin_interfaces/Time stamp
string frame_id
```

the `stamp` is the type of `builtin_interfaces/Time`, and you have to assign to it with an object of the exact type. 

``` JavaScript
let header = new Header();
let stamp = new Time({sec: 100, nanosec:100});
header.stamp = stamp;
``` 

We want to implement the code below to simplify it:

``` JavaScript
let header = new Header();
header.stamp = {sec: 100, nanosec:100};
```

qiuzhong, 2018-06-08:
Tried on `actionlib` branch, the simplified way:
```
let header = new Header();
header.stamp = {sec: 100, nanosec:100};
```
can work well.

minggangw, 2018-06-14:
Close, as already merged with `actionlib` branch

PR #355 (2018-06-07): [rosidl]Support to assign the property of the message with a plain
object

For current messages generated by rclnodejs, if you want to assign a
property of the message to a new value, the type of it must be the same
with the type of the property to be assigned, e.g.

let header = new Header();
let stamp = new Time({sec: 100, nanosec:100});
header.stamp = stamp;

This patch implemented that a plain object can be assign to the property
of the message, e.g.

let header = new Header();
header.stamp = {sec: 100, nanosec:100};

Fix #351
commit 60067a8d (2018-06-07): [rosidl]Support to assign the property of the message with a plain
object

For current messages generated by rclnodejs, if you want to assign a
property of the message to a new value, the type of it must be the same
with the type of the property to be assigned, e.g.

let header = new Header();
let stamp = new Time({sec: 100, nanosec:100});
header.stamp = stamp;

This patch implemented that a plain object can be assign to the property
of the message, e.g.

let header = new Header();
header.stamp = {sec: 100, nanosec:100};

Fix #351
closed 2018-06-14 by minggangw