RobotWebTools/rclnodejs #356: Use postinstall to generate the JavaScript messages

minggangw, 2018-06-08:
As the rclnodejs will generate the JavaScript messages based on the IDL files (.srv/.msg), the time consumption of first-time start up is usually over several seconds.

We want to move the generation of the messages to the stage of installing to accelerate the start up speed for the first time.

qiuzhong, 2018-06-08:
@minggangw , does this mean that `rclnodejs.init()` will not generate these message files any more?

minggangw, 2018-06-08:
During the installation (`npm install`), the message files will be created, and if `rclnodejs.init()` finds that the work has been done, they will not generated again which can accelerate the speed of start up.

qiuzhong, 2018-06-08:
Sure. I'm testing for it.

qiuzhong, 2018-06-08:
Just tried #357  and it worked!

PR #357 (2018-06-14): Pre-generate the JavaScript messages during installation by postinstall
To accelerate the speed of first-time start up, we better pre-generate
the messages during installation. Thus we do not have to wait the work
to be done when the module is initialized.

Fix #356
commit ead365e4 (2018-06-14): Pre-generate the JavaScript messages during installation by postinstall

To accelerate the speed of first-time start up, we better pre-generate
the messages during installation. Thus we do not have to wait the work
to be done when the module is initialized.

Fix #356
commit 619cc195 (2018-07-09): Pre-generate the JavaScript messages during installation by postinstall

To accelerate the speed of first-time start up, we better pre-generate
the messages during installation. Thus we do not have to wait the work
to be done when the module is initialized.

Fix #356
commit 6abd47f8 (2018-07-09): Pre-generate the JavaScript messages during installation by postinstall

To accelerate the speed of first-time start up, we better pre-generate
the messages during installation. Thus we do not have to wait the work
to be done when the module is initialized.

Fix #356
closed 2018-06-14 by minggangw