RobotWebTools/rclnodejs #361: [PnP]Benchmark tests plan CheckList

minggangw, 2018-06-14:
Track the Benchmark test tasks (step 1 of 3)

- [ ] Add parameters for the command line
- [ ] pre-gen when installing
- [ ]  Add Startup time test 
- [ ]  Performance of different messages
- [ ]  Auto script
- [ ] Auto report/chart generation
- [ ]  The maximum data capacity 
- [ ] Make sure every language is with best optimization

minggangw, 2018-06-14:
@hanyia @kenny-y I created this issue to track our target of the benchmark test.

hanyia, 2018-06-26:
Till now, based on our testing, about 250000KB is the maximum capacity of C++/Node.js client and 200000KB as Python(subscription/publisher samples).

minggangw, 2018-07-24:
>Auto script @hanyia
Auto report/chart generation @kenny-y

As these two items have finished and we have fixed some critical bugs during these days, hopefully we will have the release to be published recently, which will build against the latest master branch of ROS 2.0