RobotWebTools/rclnodejs #408: Got crashed when subscribe a nav_msgs/msg/Odometry message from gazebo

qiuzhong, 2018-08-15:
Tried to subscribe `nav_msgs/msg/Odometry` message in `rclnodejs` envrionments while launching `ros1_bridge` and ROS1 gazebo simulator but got crashed.

Steps to produce:
* Launch ROS1 gazebo simulator
* Launch `ros1_bridge` in ROS2 environment:
  `ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`
* Try to subscribe `nav_msgs/msg/Odometry` message: `$ node poseSub.js`, here is the code of poseSub.js: 
```
'use strict';

const rclnodejs = require('rclnodejs');

rclnodejs.init().then(() => {

  let poseNode = rclnodejs.createNode('turtlebot3_pose_node');
  let poseSub = poseNode.createSubscription('nav_msgs/msg/Odometry', '/odom', function(msg) {
    console.log(msg);
    console.log('pose msg received');
  });

  rclnodejs.spin(poseNode);
}).catch((err) => {
  console.log(err);
});

```
Then it got crashed.

qiuzhong, 2018-08-15:
@minggangw , would you please take a look ? And we can discuss with more details online.

minggangw, 2018-08-15:
@qiuzhong I am going to reproduce it locally first, any question I will liaise you offline. Thanks!

qiuzhong, 2018-08-15:
PS: I didn't encounter any crashes if I install the `rclnodejs` from github. However, `ros2-web-bridge` depends on version `0.4.1`.

minggangw, 2018-08-15:
I notice that the message, **Odometry**, has a static field(`float64[36] covariance` in PoseWithCovariance.msg).

I landed two commits which were 9cac9627edc3ad93d1a46ec2c08b47782d9d6ef3 and 9cac9627edc3ad93d1a46ec2c08b47782d9d6ef3 to fix the memory layout issues for static array in last week. So I think the crash should be caused by the static array in the message.

@qiuzhong Please double-check it and if it can not be reproduced on `develop` branch, I am going to have a bug fix release in this week, then we can get over it for rosbridge. Thanks!

qiuzhong, 2018-08-16:
Tested with commit 9cac9627edc3ad93d1a46ec2c08b47782d9d6ef3 and  no crash again.

qiuzhong, 2018-08-16:
New release `0.4.2` fixed this issue.

closed 2018-08-16 by qiuzhong