RobotWebTools/rclnodejs #440: Build error when npm install

yfzcopy0702, 2018-12-20:
My ros2 is built from source, https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
I use the step in README, when I run npm install:

CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
../src/executor.cpp:17:10: fatal error: rcl/error_handling.h: No such file or directory
 #include <rcl/error_handling.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
rclnodejs.target.mk:124: recipe for target 'Release/obj.target/rclnodejs/src/executor.o' failed
make: *** [Release/obj.target/rclnodejs/src/executor.o] Error 1
make: Leaving directory '/home/fuzhiyang/github/rclnodejs/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:254:12)
gyp ERR! System Linux 4.15.0-42-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/fuzhiyang/github/rclnodejs
gyp ERR! node -v v11.4.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rclnodejs@0.7.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.7.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

eeyang92, 2018-12-25:
I'm not too familiar with `node-gyp`, but running the `include` part of the config file:

```
// binding.gyp
...
'conditions': [
        [
          'OS=="linux"', {
            'defines': [
              'OS_LINUX'
            ],
            'cflags_cc': [
              '-std=c++14'
            ],
            'include_dirs': [
              "<!@(node -e \"console.log(process.env.COLCON_PREFIX_PATH.replace(/:/, '/include/ ') + '/include/')\")",
            ],
            'library_dirs': [
              "<!@(node -e \"console.log(process.env.COLCON_PREFIX_PATH.replace(/:/, '/lib/ ') + '/lib/')\")",
            ],
          }
        ],
...
```

```
node -e "console.log(process.env.COLCON_PREFIX_PATH.replace(/:/, '/include/ ') + '/include/')"
```

Result:
```
/home/ericy/ros2_ws/install/include/
```

Which does not exist

Under the `install` directory, it looks like there are the individual packages, such as `rcl`, `rcutils`, `rosidl_generator_c`, etc. Adding them one by one in the `include_dirs` and `library_dirs` resolves the issue - at least until it can't find the next dependency. I'm not really sure how to tell `node-gyp` to look at each subdirectory in `install` for the `include` and `lib` directories, rather than manually entering them one by one.

minggangw, 2018-12-25:
Due to the recent changes of rcl library (interfaces change mainly), I suggest you download the binary package of ROS2 from [here](https://ci.ros2.org/view/packaging/job/packaging_xenial_linux/ ) which were built before **11/20/2018.** What's more, https://github.com/RobotWebTools/rclnodejs/issues/437 is expected to fix the compiling error on the [master branch of ROS 2](https://github.com/ros2/ros2/blob/master/ros2.repos).

For the specific problem from @yfzcopy0702, please confirm you are building with flag `--merge-install` although you maybe meet a compiling error I mentioned above later :(

eeyang92, 2018-12-25:
Thank you @minggangw, I will give the binary a try.

minggangw, 2018-12-29:
@yfzcopy0702 @eeyang92 If this issue in irreproducible, I am going to close it. And please wait for the patch which will fix #437, thanks!

minggangw, 2018-12-29:
So let's close this issue, please feel free to reopen it if anything still wrong.

closed 2018-12-29 by minggangw