RobotWebTools/rclnodejs #469: Actions support in rclnodejs

minggangw, 2019-03-11:


minggangw, 2020-02-07:
I spent half a day to dig into this feature and found it's more difficult than I thought which means we need more effort on it. An official [introduction](https://index.ros.org/doc/ros2/Tutorials/Understanding-ROS2-Actions/) of Actions in ROS2

mattrichard, 2020-02-17:
Here's a breakdown of the work that needs to be done.

- [ ] Add actions to event loop (#570)
- [ ] Read action type support libraries (#571)
- [ ] Update message generation to properly handle .action files
- [ ] Add all action binding methods for rcl integration
- [ ] Add new ActionClient and ActionServer classes
- [ ] Integrate action clients/servers into node handle execution
- [ ] Unit/integration testing
- [ ] Update TypeScript typings
- [ ] Documentation and examples

PR #630 (2020-03-26): Add actions support to rclnodejs
Merges actions branch into develop. This PR should and is safe to be squashed.

Fix #469
closed 2020-03-26 by minggangw