RobotWebTools/rclnodejs #470: Performance issue

mwamp, 2019-03-14:
Hi,

I just want to report that using ros2_web_bridge saturates one CPU thread on my machine.
The likely issue is that processing power is never yielded as it should and some routines are constantly nagging to check for some event.

Here is a perf report, I produced this with very few topics; actions and services running.

  31.44%  node     libpthread-2.23.so    [.] pthread_mutex_lock
  27.91%  node     rclnodejs.node         [.] _ZN9rclnodejs8Executor3RunEPv
  20.04%  node     libpthread-2.23.so    [.] pthread_mutex_unlock
   4.46%  node     node                          [.] uv_mutex_unlock
   4.33%  node     node                          [.] uv_mutex_lock
   3.72%  node     rclnodejs.node           [.] uv_mutex_unlock@plt
   3.30%  node     rclnodejs.node           [.] uv_mutex_lock@plt
   2.31%  node     node                  [.] 0x00000000004b5b90
   2.24%  node     node                  [.] 0x00000000004b5be0

One question, is rclnodejs a wrapper on rclcpp or was is built from scratch. Wouldn't using swig or the like provide a suitable enough wrapper and easy to maintain?

minggangw, 2019-03-19:
Thanks for submitting the issue, I'd like to answer them:

>The likely issue is that processing power is never yielded as it should and some routines are >constantly nagging to check for some event.

For the unreasonable CPU usage, it should be caused by frequent calling of uv_mutex_lock/uv_mutex_unlock. The reason why it happens is that we set a fixed timeout for [rcl_wait](https://github.com/RobotWebTools/rclnodejs/blob/develop/src/executor.cpp#L132) currently. This parameter should be customized.

>One question, is rclnodejs a wrapper on rclcpp or was is built from scratch. 

No, rclnodejs is not based on rclcpp, actually all clients of ROS2, including C++, Python and Node.js are based on [rcl](https://github.com/ros2/rcl) which is a C library, not platform related. You could consider rclnodejs is kind of wrapper, meanwhile, implement some platform specific features, like thread.

>Wouldn't using swig or the like provide a suitable enough wrapper and easy to maintain?

I'm not familiar with swig, not sure it can meet the requirement to implement the functions.

Finally, you are welcome to fix any problem you found, thanks again!

minggangw, 2019-07-05:
This is a little late to come, but we did it finally. If you are still using it, please have a try to see if the performance changes. Thanks a lot!

closed 2019-07-05 by minggangw

mwamp, 2019-07-05:
Great, I'll check it out as soon as I possible. 
Thanks a lot.

mwamp, 2019-07-15:
Hi,
Unfortunately, I am unable to run tests right now. I am hit with a segfault (the very dark, no exception raised kind) as soon as rclnodejs is launched.

I am running dashing so maybe the reason is that I would need the nightly ros build?

minggangw, 2019-07-15:
I didn't verify it on Dashing actually, as the code on `develop` branch will be tested on CIs by running unit tests. So I am going to double check it on the Dashing release. Thanks for your feedback.

mwamp, 2019-07-15:
Actually, it looks like there are issues with NodeJS on my side, there shouldn't be anything wrong with this code.

minggangw, 2019-07-16:
My previous description is not clear, I mean the `develop` branch of rclnodejs will be tested on the nightly ROS2 package through travis-ci. The Node.js version is the latest stable release usually.

PR #472 (2019-07-05): Add timeout parameter for rcl_wait functioin
Before this patch, the timeout used for the rcl_wait function is hard
coded which is 10ms. Apparently, this is not reasonable, because it
imposes unnecessary workload on the CPU (We need to lock/unlock the
resources shared between the main thread and the background thread
frequently). Please check the detailed record in the issue.

This patch adds the timeout parameter which developer can customize the
value when starting a node. The default value now is 10ms, negtive value
will block the thread until there is an event.

So you can change your JS co [... cut]
commit 54a6c1f3 (2019-07-05): Add timeout parameter for rcl_wait function

Before this patch, the timeout used for the rcl_wait function is hard
coded which is 10ms. Apparently, this is not reasonable, because it
imposes unnecessary workload on the CPU (We need to lock/unlock the
resources shared between the main thread and the background thread
frequently). Please check the detailed record in the issue.

This patch adds the timeout parameter which developer can customize the
value when starting a node. The default value now is 10ms, negtive value
will block the thread until there is an event.

So you can change your JS co [... cut]