RobotWebTools/rclnodejs #474: COLCON_PREFIX_PATH vs AMENT_PREFIX_PATH

robotpilot, 2019-03-21:
The [binding.gyp](https://github.com/RobotWebTools/rclnodejs/blob/develop/binding.gyp) file uses `COLCON_PREFIX_PATH` instead of `AMENT_PREFIX_PATH`. However, [README.md](https://github.com/RobotWebTools/rclnodejs/blob/develop/README.md)(Build Module section) describes `AMENT_PREFIX_PATH`. Which is the correct explanation?

minggangw, 2019-03-22:
Oops, this is historical issue caused by the toolchain transition from ament to colcon. Currently, both of the two varieties are supported by ROS 2, but I suggest to use the colcon.

@robotpilot thank for pointing it out, I will unify this information in the README file.

robotpilot, 2019-03-22:
LGTM!
Thanks, @minggangw :)

PR #475 (2019-03-22): Use COLCON_PREFIX_PATH to replace AMENT_PREFIX_PATH
Fix #474
commit d117fffa (2019-03-22): Use COLCON_PREFIX_PATH to replace AMENT_PREFIX_PATH

Fix #474
closed 2019-03-22 by robotpilot