RobotWebTools/rclnodejs #480: Error when building in docker container

samiamlabs, 2019-04-09:
Hi, 

I'm having some trouble getting this to build in a docker container.  I get this error when running ```npm install --unsafe-perm --verbose``` as root in the container (after sourcing ROS2):

../src/executor.cpp:17:10: fatal error: rcl/error_handling.h: No such file or directory
 #include <rcl/error_handling.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
rclnodejs.target.mk:120: recipe for target 'Release/obj.target/rclnodejs/src/executor.o' failed
make: *** [Release/obj.target/rclnodejs/src/executor.o] Error 1
make: Leaving directory '/opt/rclnodejs/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2

I tried both in a container built from the Dockerfile in this repository and using my own image. It seems to be working fine on linux outside the container.

minggangw, 2019-04-10:
Hi @samiamlabs, thanks for submitting this issue.

Currently, the travis-ci of rclnodejs is also running in docker, where the Dockerfile is https://github.com/RobotWebTools/rclnodejs/blob/develop/Dockerfile, and you can check the building result https://travis-ci.org/RobotWebTools/rclnodejs

minggangw, 2019-04-10:
Oh, I missed the information that you have tried with the Dockerfile in this repo, sorry. The error message indicates that the `include_dir` of the headers for ROS2 cannot be found, have you had executed something like `. ~/ros2_install/ros2-linux/setup.bash` already?

samiamlabs, 2019-04-10:
The docker image has that sourcing command in .bashrc so that is not the problem.

Got it to work now. I had to use ```./scripts/build.sh``` instead of  ```npm install --unsafe-perm``` to build. Not really sure why.

minggangw, 2019-04-11:
It is a little weird. Last time I met this kind of problem, the reason was that I built the ROS2 from source **without** the flag `--merge-install` (with this flag, it installs all packages into a combined directory, rather than each package in a separate subdirectory)

Anyway, as it has worked, I am going to close this issue. You are welcome to give any feedback if you find the root cause in the future. Thanks!

closed 2019-04-11 by minggangw