RobotWebTools/rclnodejs #481: Create Interfaces from Strings

mwamp, 2019-04-18:
From the way ros2web-bridge is made it looks like interfaces can be loaded using strings encoded according to the format : ``{std_msd | ... }/{msg | srv}/{String | ...}``.
It looks like a function was written in this direction in the interface loader but was never called. For instance [here](https://github.com/mwamp/rclnodejs/blob/aefdc7195f7978620693d42768282985218468c4/lib/node.js#L248) a condition checks for a string but calls the generic loadInterface instead of loadInterfaceByString.

By the way AFAIK these functionnalities are not referenced anywhere? It would be nice to have some kind of specification/documentation for this so we know in the future why such choices were made. As I see it this is part of the api rclnodejs exposes yes?

IMO the whole idea of formatting the string as above doesn't feel very ROSish or JSish but I am no expert.

minggangw, 2019-04-19:
Would you please write a code snippet to demonstrate the aim you want to achieve after the PR #482, thanks!

mwamp, 2019-04-19:
Sorry it seems this rewrite just helped me expose errors due to not yet generated idl_mgs/srv and I afterwards thought it was necessary when it is not.
My bad

closed 2019-04-19 by mwamp