RobotWebTools/rclnodejs #484: ROS package

mwamp, 2019-04-18:
This is not a ros package. Maybe it should be and serve as an example of what a JS ros2 package looks like.

What's more AFAIU adapters are also here to facilitate ROS integration. I would expect to be able to build a package and write something like ``<depend>rclnodejs</depend>`` and then call my JS package in a launch file using the launch system (which would be a bit heavy since we would then expect to write smthing like colcon_js to integrate with the build system).

This might seem a bit useless for packages like ros3DJS which are more "UI components" (and even then maybe not) but definitely not for ro2-web-bridge which in ros1 was in python if I am not mistaken.

If no one really has time to really push this project (I don't) maybe we should still have a letter of intention with the steps to build an adapter which really mimics rclcpp/rclpy

minggangw, 2019-04-19:
Thanks for your suggestion! Actually, we considered how to integrate the `rclnodejs` into the toolchain of ROS2 at the beginning of this project, and as you said we have to change the current `colcon` to adapt to the Node.js packages. 

But I believe people who want to use this package tend to be the JavaScript developers, they feel more comfortable if using `npm` because it's the toolchain of Node.js eco-system. Meanwhile, if the `rclnodejs`  can be supported by `colcon`, it would be better and developers can have more choices.

Finally, we don't have the extra effort on integration work, and if you have some constructive suggestion,  we can discuss it here, thanks!

minggangw, 2020-01-29:
Hi @mwamp here comes good news 😄 now @wayneparrott is working actively on what you need (see https://github.com/ros2/ros2cli/issues/427)