RobotWebTools/rclnodejs #492: QoS - API and implementation for Liveliness and Deadline event callbacks

minggangw, 2019-05-27:


minggangw, 2020-02-25:
Continued from #498 

>for this thing I need to add one more paramter in the methods which is used for creating Publisher/Subsciber, so I m asking if need to embed this new parameter inside options object or add just add a new paramter field.
What guidelines is followed I dont know

We suggest aligning with interfaces of rclpy, there is no guideline for designing an interface

>what is the use of timerCallback(),I have seen that at the end this function calls rcl_timer.c but cant understand what is its role

`timer.callback()` will trigger the callback function passed when creating this timer

caelinsutch, 2020-07-13:
Is this taken?

minggangw, 2020-07-13:
No, it' not. Please feel free to take it if you are interested in.

minggangw, 2025-06-06:
Fixed by #1159

closed 2025-06-06 by minggangw