RobotWebTools/rclnodejs #510: build fail on ubuntu with dashing

koonpeng, 2019-09-11:
When generating javascript messages, I got the following error message
```
Start to generate the JavaScript messages...
Caught error: Error: EACCES: permission denied, mkdir '/opt/ros/dashing/share/tf2_msgs/action/msg'
```
It looks like the current method to support action creates the `.msg` files from the `.action` files and uses them to generate the messages. This fails in ubuntu as its trying to create `.msg` files in a directory owned by root.

The result is that some messages are not generated, particularly packages that include both `.action` and `.msg` files, e.g. `example_interfaces`. This also causes some tests to fail as they rely on `example_interfaces`.

minggangw, 2019-09-11:
I see, do you install dasing via Debian Packages? As the actionlib has been [deprecated](https://github.com/RobotWebTools/rclnodejs#experimental---deprecated), I think we can remove these tests from the current test set.

koonpeng, 2019-09-12:
Yes, they are installed using `apt`. For now I am working around it by commenting out some lines to disable actions too.

wayneparrott, 2019-12-05:
Any thoughts on how to address the EACCESS issue on linux from the generate_messages script?

In eloquent the dirs that the script is attempting to write into are:
```
/opt/ros/eloquent/share/
  action_tutorials_interfaces/actions
  example_interfaces/actions
  tf2_msgs/actions
```
Brainstorm how to proceed:
1) A bad hack is to make the dirs above writable.
2) Disable generating action messages until we have a proper solution.
3) Generate msg files into the project and process them in the overlay. 
Thoughts?

minggangw, 2019-12-05:
The implementation of [actionlib](http://wiki.ros.org/actionlib) of `rclnodejs` is prior to that of `rcl`, so me have marked the actionlib as [deprecated](https://github.com/RobotWebTools/rclnodejs#experimental---deprecated) (we haven't achieved to wrap the actionlib related functions from `rcl` and exposed them). I prefer to remove the generation for `.action` because they become useless now, which will eliminate the current error you meet on Linux.

I am going to submit a PR to remove it, thanks

PR #542 (2019-12-06): Disable the generation of actionlib
Fix #510
commit ac497ef1 (2019-12-06): Disable the generation of actionlib

Fix #510
closed 2019-12-06 by minggangw