RobotWebTools/rclnodejs #511: maximum recursion error when running in jest

koonpeng, 2019-09-13:
There is a bug in `ref` which throws maximum recursion error when the module is loaded multiple times. Normally this error wouldn't happen because of the nodejs require cache. However this breaks rclnodejs when running in jest because jest ignores the require cache and ref is loaded multiple times, triggering the bug.

It can be reproduced with

a.test.js:
```
const rclnodejs = require('rclnodejs');

it('a', async () => {
  await rclnodejs.init();
  rclnodejs.shutdown();
});
```
b.test.js:
```
const rclnodejs = require('rclnodejs');

it('b', async () => {
  await rclnodejs.init();
  rclnodejs.shutdown();
});
```
`npx jest <test-dir>`

running the test, I got the error
```
 PASS  test/a.test.js
 FAIL  test/b.test.js
  ● b

    RangeError: Maximum call stack size exceeded

      at debug (../node_modules/ref/node_modules/debug/src/debug.js:1:1)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:746:3)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)
      at Object.writePointer [as _writePointer] (../node_modules/ref/lib/ref.js:747:11)

Test Suites: 1 failed, 1 passed, 2 total
Tests:       1 failed, 1 passed, 2 total
Snapshots:   0 total
Time:        0.365s, estimated 6s
Ran all test suites matching /.\/test/i.
```

There is a PR already submitted to ref that fix this but it hasn't been merged for a long time and ref doesn't seem to be maintained anymore.

minggangw, 2019-09-14:
Thanks for your findings! Through your description, I understand this problem happens only under the jtest framework, right? 

The current test framework used by `rclnodjes` is [mocha](https://www.npmjs.com/package/mocha), and I haven't seen this problem yet, which means it uses the cache I think. Considering what you said, this problem is not triggered by the normal nodejs usage. Meanwhile, I checked the [ref](https://www.npmjs.com/package/ref) and it seems that it has already been out of maintenance for about 2 years, so we cannot expect a fix of that problem :( 

To sum up, this is not a fatal error for most developers, I think we can add a note to the [Run Unit Test](https://github.com/RobotWebTools/rclnodejs#run-unit-test) section to mention not to use jtest framework for testing. If you agree, would you please add it as you have made a deep investigation, thanks!

koonpeng, 2019-09-16:
Yes, this only occurs in jest, mocha runs fine for me. I have been trying out workarounds using mocks as shown in this blog http://www.petecorey.com/blog/2018/11/05/bending-jest-to-our-will-caching-modules-across-tests/ but I wasn't able to get it working.

I think a possible solution is to fork `ref` and have `rclnodejs` install the forked repo by using shrinkwrap, or publish new versions of `ref*` and have `rclnodejs` use them. Those are pretty drastic solutions and I don't know if its a good idea. For now I will just add a note that jest will not work.

minggangw, 2019-09-17:
>I think a possible solution is to fork ref and have rclnodejs install the forked repo by using shrinkwrap, or publish new versions of ref* and have rclnodejs use them. Those are pretty drastic solutions and I don't know if its a good idea. For now I will just add a note that jest will not work.

Yes, I agree that forking `ref` is not necessary as it only happens in jtest. So I suggest keeping it as it is. Thanks!

closed 2019-09-25 by koonpeng